Jasonchenlijian/FastBle #23: 设备返回值

wdh-1025, 2016-11-28:
你好，请问写入数据后如何接收设备的返回值呢？

Jasonchenlijian, 2016-11-29:
notifyDevice

cwfzc, 2016-11-29:
你好，我在设备连接的时候，如果连接成功就notifyDevice接收参数，但是设备需要配对，配对成功了以后，有一次连接成功，然后notifyDevice就执行了两次，结果收到的数据打印的时候，打印两次，间隔不到2毫秒，请问这样要怎么处理呢？

Sunny22100486, 2016-12-26:
你好，我在使用的时候每次连接之后，然后再返回上个页面（没有主动断开连接），再次连接的时候BleGattCallback执行次数会累加，这是什么原因呢

Jasonchenlijian, 2016-12-26:
bleManager.stopListenCharacterCallback

Sunny22100486, 2016-12-26:
这个这是移除某个特征值回调监听的吧，会不会是我每次连接的时候是不是都new 了一个BleGattCallback呢？

Jasonchenlijian, 2016-12-26:
每次都new没有问题，但记住用完之后调用bleManager.stopNotify。 否则建议不要new

Sunny22100486, 2016-12-26:
我现在想，每次连接之后就打开notify，之后就不打开notify 了，其实我是希望可以一直不关闭notify

发自网易邮箱大师
在2016年12月26日 21:23，陈利健 写道:

每次都new没有问题，但记住用完之后调用bleManager.stopNotify。 否则建议不要new

Jasonchenlijian, 2016-12-26:
那你就构造一个BleGattCallback的子类对象，一直用它就好。

Slj7663, 2017-01-18:
蓝牙连接成功之后由于某些原因断开了，再次在同一个页面调用连接，成功时会调用两次连接回调里的onServicesDiscovered，我在这里面打开了notify，所以notify也会调用两次，所以会出现接收到重复的消息，是因为我这边用法的错误，还是其他原因呢？

closed 2017-03-15 by Jasonchenlijian