Jasonchenlijian/FastBle #32: Android6.0掃描不到裝置

imcloudwu, 2017-02-03:
已經在Manifest添加以下權限

    <uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />

仍然掃描不到裝置,請問能成功運行在6.0以上的解法?

littleKun, 2017-02-07:
一样的问题,没有解决

Tommy-Fen, 2017-02-24:
对应的两个硬件功能的声明加上试一下
```
<uses-feature android:name="android.hardware.location.gps" />
uses-feature android:name="android.hardware.location.network" />
```

iYoullloveGitHub, 2017-02-27:
private void checkBluetoothPermission() {
        if (Build.VERSION.SDK_INT >= 23) {
            //校验是否已具有模糊定位权限
            if (ContextCompat.checkSelfPermission(MainActivity.this,
                    Manifest.permission.ACCESS_COARSE_LOCATION)
                    != PackageManager.PERMISSION_GRANTED) {
                //申请模糊定位权限 requestCode =1
                ActivityCompat.requestPermissions(MainActivity.this,
                        new String[]{Manifest.permission.ACCESS_COARSE_LOCATION}, 1);
            }
            //校验是否已具有精确定位权限
            if (ContextCompat.checkSelfPermission(MainActivity.this,
                    Manifest.permission.ACCESS_FINE_LOCATION)
                    != PackageManager.PERMISSION_GRANTED) {
                //申请精确定位权限  requestCode =2
                ActivityCompat.requestPermissions(MainActivity.this,
                        new String[]{Manifest.permission.ACCESS_FINE_LOCATION}, 2);
            }
        }
    }

加上这段代码试试,    6.0 以后需要动态申请权限

closed 2017-03-15 by Jasonchenlijian