Jasonchenlijian/FastBle #42: 连接成功以后马上就会主动的断开

JiWenBo, 2017-03-21:
已经显示21 15:54:12.818 25288-25300/com.sita.fastbleactivity E/1000000111000: 蓝牙获取服务成功
03-21 15:54:12.818 25288-25300/com.sita.fastbleactivity D/BluetoothDevice: mAddress: A8:1B:6A:85:A8:66
03-21 15:54:12.818 25288-25300/com.sita.fastbleactivity D/BluetoothDevice: mAddress: A8:1B:6A:85:A8:66
03-21 15:54:12.819 25288-25300/com.sita.fastbleactivity D/BluetoothDevice: mAddress: A8:1B:6A:85:A8:66
03-21 15:54:12.819 25288-25300/com.sita.fastbleactivity E/1000000111000: 蓝牙特征值获取成功
03-21 15:54:12.819 25288-25300/com.sita.fastbleactivity D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000ffe1-0000-1000-8000-00805f9b34fb enable: true
03-21 15:54:12.819 25288-25300/com.sita.fastbleactivity D/BluetoothDevice: mAddress: A8:1B:6A:85:A8:66
03-21 15:54:12.822 25288-25288/com.sita.fastbleactivity D/BluetoothDevice: mAddress: A8:1B:6A:85:A8:66
03-21 15:54:12.823 25288-25288/com.sita.fastbleactivity D/BleConnector: 0000ffe1-0000-1000-8000-00805f9b34fb
                                                                        characteristic.getProperties():24
                                                                        characteristic.getValue(): null
                                                                        characteristic write bytes: [24, 24, 13, 13, 1, 0, 0, 0, 0, 0, 0, 13, 10]
                                                                        hex: 18180d0d010000000000000d0a
E/FaseBlueLog: 连接失败或连接中断：201
03-21 15:54:12.998 25288-25301/com.sita.fastbleactivity E/BleExceptionHandler: Gatt Exception Occurred!

Xanthuim, 2017-03-23:
其实有时候就是这样的，我自己写的时候有时候也会掉。可能与设备的当时的环境有关。

JiWenBo, 2017-04-01:
我连接成功了，断开蓝牙以后再打开蓝牙，然后再次连接成功以后，发送一条指令过去以后就有两条蓝牙消息回调，这是为什么啊？

Jasonchenlijian, 2017-06-29:
新版已修复了此处问题，可以尝试调试您的设备。

closed 2017-11-29 by Jasonchenlijian