Jasonchenlijian/FastBle #51: 你好，startNotify方法执行了，characteristic和service的uuid也写入了，但是通知无返回值是什么情况

lioulei1317, 2017-04-01:
04-01 10:00:47.703 26119-26119/com.bdk.blesample I/OperateActivity: startNotify
04-01 10:00:47.704 26119-26119/com.bdk.blesample W/BleConnector: characteristic.getProperties():58
04-01 10:00:47.704 26119-26119/com.bdk.blesample D/BluetoothGatt: setCharacteristicNotification() - uuid: bef8d6c9-9c21-4c9e-b632-bd58c1009f9f enable: true
04-01 10:00:47.706 26119-26119/com.bdk.blesample D/BleConnector: setCharacteristicNotification: true
                                                                 success: true
                                                                 characteristic.getUuid(): bef8d6c9-9c21-4c9e-b632-bd58c1009f9f

lizhongze123, 2017-04-01:
这代码里面的是你startNotify之后，这里的onSuccess回调只是接收到notify的成功回调，不是设置之后成功的回调，如果你是要设置成功之后的回调，需要你自己重写一下handleCharacteristicNotificationCallback这个方法里的onDescriptorWrite这个方法的，不知道你问的是不是这个问题。才疏学浅，莫怪。

lioulei1317, 2017-04-01:
@lizhongze123 我要的就是接收仪器返回的notify，service的uuid和characteristic的uuid确定都是对的，在我的项目中是可用的，而且刚开始运行项目是有返回值的，后来用就一直不返回了

lizhongze123, 2017-04-01:
这个很难判断错误在哪里。或者你用其他蓝牙工具看可不可以接收到，如果可以，那就不是UUID的问题了，那你就要看看你的代码了

lioulei1317, 2017-04-01:
@lizhongze123 好的，谢谢，我再看看

lioulei1317, 2017-04-02:
@lizhongze123 上次的问题解决了，遇到新的问题，如果需要连续监听多个notify的uuid，该怎么做？

androidxu, 2017-04-05:
@lioulei1317 请问你是怎么解决的？

lioulei1317, 2017-04-07:
@androidxu 要有一个时间间隔，我是监听和写入隔了100ms，如果写在一起不间隔就不返回

closed 2017-05-01 by Jasonchenlijian