Jasonchenlijian/FastBle #52: 调用writeDevice写入总是失败？

TearBack, 2017-04-05:
RT，这是调用writeDevice写入时，日志打印的信息：
D/BleConnector: 8bc6761e-4052-8eba-a7c992ba05d5
                characteristic.getProperties():42
                characteristic.getValue(): null
                characteristic write bytes: [1]
                hex: 01
D/BluetoothGatt: writeCharacteristic() - uuid: 8bc6761e-a8ba-4052-8eba-a7c992ba05d5
D/BluetoothGatt: writeCharacteristic: mDeviceBusy = true, and return false
D/BleConnector: initiated: false
而我自己使用原生的写入方法就可以写入成功，Ble设备终端也收到了指令。
mCharacterCmd.setValue(cmdBytes);
boolean isWriteSucc = mBluetoothGatt.writeCharacteristic(mCharacterCmd);
为什么会酱紫？

TearBack, 2017-04-05:
补充一下，发现个问题：
mCharacterCmd.setValue(cmdBytes);
boolean isWriteSucc = mBluetoothGatt.writeCharacteristic(mCharacterCmd);
我发现，调用这个写入，放在子线程才可以成功，如果放在主线程，就会
D/BluetoothGatt: writeCharacteristic: mDeviceBusy = true, and return false
有这个提示而失败。

suhuangqiang, 2017-12-19:
请问解决了吗

Jasonchenlijian, 2018-01-11:
此问题在新版本上已做了优化。

closed 2018-01-11 by Jasonchenlijian