Jasonchenlijian/FastBle #68: 问题好多啊，要哭瞎了，早知道我自己写了，没时间了

Bter, 2017-05-11:
问题1：isConnected()和isConnectingOrConnected()状态判断不稳定，假设我已经连接了，然后我就扫描，这时状态就不对了。
问题2：closeBluetoothGatt()有问题，有时无法回调当你要手动断开时调用disconnect方法，然后在onConnectionStateChange的状态回调中调用close方法。如果在disconnect后直接调用close，会导致不走onConnectionStateChange这里的。而且有时还会报异常。还有一个原因没回调可能是因为提前清除了所有回调。
问题3：其实这个问题算是疑惑，看作者的代码想必好像只支持单连接，但每次write数据的时候都增加回调监听（用了HashMap保存），但write之后没看到清除该回调的地方（那么无数次write后，那对象不是越来越多？），不明白为何，求解释。（我的思路：既然是单连接那么只设置一个write监听不就行了？）
其他问题有待大家发现......
慎重提醒，谨慎使用.......

zzsakurazz, 2017-05-12:
第二个问题我也发现了，感觉这个轮子还需要完善和修改

w0080626, 2017-05-24:
第二个问题我通过在他的BleBluetooth类加了个disconnect的方法
`public void disconnect() {
        if (bluetoothGatt != null) {
            bluetoothGatt.disconnect();
        }
    }`
实现的。 应该可以解决你的问题。 同时希望作者加入到库里。

Jasonchenlijian, 2017-06-29:
新版本已解决了相关问题，或者对部分API的使用进行了详细说明，感谢建议。

closed 2017-06-29 by Jasonchenlijian