Jasonchenlijian/FastBle #70: 开启多个notify，只有第一个起作用

LT13S, 2017-05-18:
请问开启多个notify，为啥只有第一个起作用？

Jasonchenlijian, 2017-05-18:
针对同一个特征，多次notify，只有最后一个起作用；针对不同特征，分别notify，每一个都会起作用。

LT13S, 2017-05-18:
没有，在同一fragment里面同时开启多个特征通知，只有第一个会回调 ，其余不起作用

Jasonchenlijian, 2017-05-18:
我不太清楚的情况，可否show一下片段代码。另外可以关注一下你每一个notify方法的返回值，代表你是否真的notify成功了。

LT13S, 2017-05-18:
mBluetoothService.notify(UUID_SERVCE,characteristic.getUuid().toString(), new BleCharacterCallback() {
                            @Override
                            public void onSuccess(final BluetoothGattCharacteristic characteristic) {
                                int value = Integer.valueOf(HexUtil.encodeHexStr(characteristic.getValue()),16) ;
                                Log.v("value11",String.format("%.2s",(float)value/(float)4)+"℃"+"   "+value +" "+HexUtil.encodeHexStr(characteristic.getValue()));
                                final String value1 = String.format("%.1f",(((float)value/(float)4-32)/1.8))+"℃";
                                Message message = new Message();
                                Bundle bundle = new Bundle();
                                bundle.putString("htm",value1);
                                message.setData(bundle);
                                handler.sendMessage(message);
                            }
                            @Override
                            public void onFailure(BleException exception) {
                                mBluetoothService.stopNotify(UUID_SERVCE,characteristic.getUuid().toString());
                            }
                        });

            mBluetoothService.notify(UUID_SERVCE, characteristic1.getUuid().toString(), new BleCharacterCallback() {
                        @Override
                        public void onSuccess(BluetoothGattCharacteristic characteristic) {
                            mBluetoothService.getGatt().setCharacteristicNotification(characteristic, true);
                            final int value = Integer.valueOf(HexUtil.encodeHexStr(characteristic.getValue()), 16);
                            Log.v("value11", HexUtil.encodeHexStr(characteristic.getValue()) + "   " + HexUtil.encodeHex(characteristic.getValue()).toString());
                            Message message = new Message();
                            Bundle bundle = new Bundle();
                            bundle.putString("gaungzhao", value + "Lx");
                            message.setData(bundle);
                            handler.sendMessage(message);
                        }

                        @Override
                        public void onFailure(BleException exception) {
                            getActivity().runOnUiThread(new Runnable() {
                                @Override
                                public void run() {
                                    guangzhaoTextView.setText("无法显示数据");
                                }
                            });
                        }
                    });
只有第一个会回调

LT13S, 2017-05-18:
mBluetoothService.notify(UUID_SERVCE,characteristic.getUuid().toString(), new BleCharacterCallback() {
                            @Override
                            public void onSuccess(final BluetoothGattCharacteristic characteristic) {
                                int value = Integer.valueOf(HexUtil.encodeHexStr(characteristic.getValue()),16) ;
                                Log.v("value11",String.format("%.2s",(float)value/(float)4)+"℃"+"   "+value +" "+HexUtil.encodeHexStr(characteristic.getValue()));
                                final String value1 = String.format("%.1f",(((float)value/(float)4-32)/1.8))+"℃";
                                Message message = new Message();
                                Bundle bundle = new Bundle();
                                bundle.putString("htm",value1);
                                message.setData(bundle);
                                handler.sendMessage(message);
                            }
                            @Override
                            public void onFailure(BleException exception) {
                                mBluetoothService.stopNotify(UUID_SERVCE,characteristic.getUuid().toString());
                            }
                        });

            mBluetoothService.notify(UUID_SERVCE, characteristic1.getUuid().toString(), new BleCharacterCallback() {
                        @Override
                        public void onSuccess(BluetoothGattCharacteristic characteristic) {
                            mBluetoothService.getGatt().setCharacteristicNotification(characteristic, true);
                            final int value = Integer.valueOf(HexUtil.encodeHexStr(characteristic.getValue()), 16);
                            Log.v("value11", HexUtil.encodeHexStr(characteristic.getValue()) + "   " + HexUtil.encodeHex(characteristic.getValue()).toString());
                            Message message = new Message();
                            Bundle bundle = new Bundle();
                            bundle.putString("gaungzhao", value + "Lx");
                            message.setData(bundle);
                            handler.sendMessage(message);
                        }

                        @Override
                        public void onFailure(BleException exception) {
                            getActivity().runOnUiThread(new Runnable() {
                                @Override
                                public void run() {
                                    guangzhaoTextView.setText("无法显示数据");
                                }
                            });
                        }
                    });

LT13S, 2017-05-18:
boolean bool = mBluetoothService.notify(UUID_SERVCE,characteristic.getUuid().toString(), new BleCharacterCallback() {
                            @Override
                            public void onSuccess(final BluetoothGattCharacteristic characteristic) {
                                int value = Integer.valueOf(HexUtil.encodeHexStr(characteristic.getValue()),16) ;
                                Log.v("value11",String.format("%.2s",(float)value/(float)4)+"℃"+"   "+value +" "+HexUtil.encodeHexStr(characteristic.getValue()));
                                final String value1 = String.format("%.1f",(((float)value/(float)4-32)/1.8))+"℃";
                                Message message = new Message();
                                Bundle bundle = new Bundle();
                                bundle.putString("htm",value1);
                                message.setData(bundle);
                                handler.sendMessage(message);
                            }
                            @Override
                            public void onFailure(BleException exception) {
                                mBluetoothService.stopNotify(UUID_SERVCE,characteristic.getUuid().toString());
                            }
                        });

    boolean bl = mBluetoothService.notify(UUID_SERVCE, characteristic1.getUuid().toString(), new BleCharacterCallback() {
                        @Override
                        public void onSuccess(BluetoothGattCharacteristic characteristic) {
                            mBluetoothService.getGatt().setCharacteristicNotification(characteristic, true);
                            final int value = Integer.valueOf(HexUtil.encodeHexStr(characteristic.getValue()), 16);
                            Log.v("value11", HexUtil.encodeHexStr(characteristic.getValue()) + "   " + HexUtil.encodeHex(characteristic.getValue()).toString());
                            Message message = new Message();
                            Bundle bundle = new Bundle();
                            bundle.putString("gaungzhao", value + "Lx");
                            message.setData(bundle);
                            handler.sendMessage(message);
                        }

                        @Override
                        public void onFailure(BleException exception) {
                            getActivity().runOnUiThread(new Runnable() {
                                @Override
                                public void run() {
                                    guangzhaoTextView.setText("无法显示数据");
                                }
                            });
                        }
                    });

Jasonchenlijian, 2017-05-18:
boolean bl的值是？如果是false，尝试一下两次notify之间间隔100ms。

LT13S, 2017-05-18:
new Thread(new Runnable() {
                @Override
                public void run() {
                    try {
                        Thread.sleep(100);
                    }catch (InterruptedException e) {
                   Log.e("error",e.toString());
                    }finally {
                        boolean bl = mBluetoothService.notify(UUID_SERVCE, characteristic1.getUuid().toString(), new BleCharacterCallback() {
                            @Override
                            public void onSuccess(BluetoothGattCharacteristic characteristic) {
                                mBluetoothService.getGatt().setCharacteristicNotification(characteristic, true);
                                final int value = Integer.valueOf(HexUtil.encodeHexStr(characteristic.getValue()), 16);
                                Log.v("value11", HexUtil.encodeHexStr(characteristic.getValue()) + "   " + HexUtil.encodeHex(characteristic.getValue()).toString());
                                Message message = new Message();
                                Bundle bundle = new Bundle();
                                bundle.putString("gaungzhao", value + "Lx");
                                message.setData(bundle);
                                handler.sendMessage(message);
                            }
                            @Override
                            public void onFailure(BleException exception) {
                                getActivity().runOnUiThread(new Runnable() {
                                    @Override
                                    public void run() {
                                        guangzhaoTextView.setText("无法显示数据");
                                    }
                                });
                            }
                        });
                    }
                }
            }).start();  
第二个是FALSE，但是我用延时会直接崩溃掉

Jasonchenlijian, 2017-05-19:
确保在主线程调用notify方法

shawnelcle, 2017-05-19:
借楼问个问题，我首先开启notify接收数据，然后写入一条数据过后（成功返回）。就收不到数据了。是我操作不对么？调用了blemanage的stopListenCharacterCallback方法后再重新notify也不行

Jasonchenlijian, 2017-05-19:
是否可以贴一下代码？

shawnelcle, 2017-05-19:
BleCharacterCallback characterCallback = new BleCharacterCallback() {
        @Override
        public void onSuccess(BluetoothGattCharacteristic characteristic) {
            byte[] bluetoothdata = characteristic.getValue();
            Logger.d("on notify:"+ HexUtil.encodeHexStr(bluetoothdata));
            sendBroadCast(bluetoothdata);
        }

        @Override
        public void onFailure(BleException exception) {
        }
    };

代码很简单
先nofity，然后write了一次，能收到write后的第一条数据。后面的就没了。onCharactoristicChanged是调用了的。上面是我自己出了问题，“调用了blemanage的stopListenCharacterCallback方法后再重新notify也不行”这个是我的问题。

shawnelcle, 2017-05-19:
我又试了一次
BleCharacterCallback writeCallback = new BleCharacterCallback() {
        @Override
        public void onSuccess(BluetoothGattCharacteristic characteristic) {
            byte[] bluetoothdata = characteristic.getValue();
            Logger.d("on write notify:"+ HexUtil.encodeHexStr(bluetoothdata));
            stopListennerCharactorCallback();
            startNotify();
        }

        @Override
        public void onFailure(BleException exception) {

        }
    };
write回调后这样执行。
public void stopListennerCharactorCallback() {
        mBluetoothService.removeBluetoothGattCallback(UUID_NOTIFY);
    }

I/FastBle: BleGattCallback：onCharacteristicChanged 
这个回调是有的

Awaken-ing, 2017-05-25:
你好，虽然可能已经来不及完成我的项目，不过我还是想弄清楚这里面的问题，获取到远程设备的MAC地址后，我开始实例化蓝牙Gatt，进行蓝牙的连接操作，确实连接成功，也读取到蓝牙的rssi了，但是下一步想读取蓝牙数据，开始实例化蓝牙service和特征值时候，就开始程序崩死了。准确的说是实例化蓝牙的特征值时候开始崩死了。。大神可有什么建议么.

Awaken-ing, 2017-05-25:
下面是我按照对BLE api的理解，自己写的代码。。下面实例化特征值后，会直接闪退，logcat是load: id=gralloc != hmi->id=gralloc
网上说是配置文件的权限问题，可是该给的权限我都给了，对于Android6.0之后的权限处理我也做代码处理了。。对了，实例化蓝牙的Gattservice和特征值，我是把特征值写死了，因为我是根据你们demo apk，事先查看了相对于的UUID。但是实际上运行会闪退，搜了很久了，卡这里快2天了。。

Recieve_button.setOnClickListener(new OnClickListener() {
			@Override
			public void onClick(View v) {
				Intent mIntent = getIntent();
				String Address =	mIntent.getStringExtra("BleAddress");
				Log.i("Adder", "Mintent: "+Address);
				BluetoothDevice device =  mBlueAdapter.getRemoteDevice(Address);//实例化远程蓝牙对象
				mGatt = device.connectGatt(MainActivity.this,false,mGattCallback);
				BluetoothGattService tempService = mGatt.getService(UUID.fromString("0000ffe0-0000-1000-8000-00805f9b34fb"));
				//可以实例化蓝牙服务。但是不可以拿到特征值
//				BluetoothGattCharacteristic mChar =	tempService.getCharacteristic(UUID.fromString("0000ffe1-0000-1000-8000-00805f9b34fb"));
//				boolean ChFlag =  mGatt.setCharacteristicNotification(mChar,true);
//				Log.e("Rs", "onClick: "+ChFlag );
				String intentAction;
				if( mGatt.connect()) {
					Toast.makeText(MainActivity.this,"连接成功",Toast.LENGTH_LONG).show();
//					String ss =	tempService.getUuid().toString();
//					Log.e("Rs", "onClick: "+ss);

				}else
				{
					intentAction =  "Devices_DISCONNECTED";
					Log.i("status", "onConnectionStateChange: "+intentAction+"已经断开连接。。。");

				}
				if(mGatt.readRemoteRssi()){

					Toast.makeText(MainActivity.this,"读取RSSI成功。。。",Toast.LENGTH_LONG).show();
					Log.e("Rs", "rssi: "+"db。。。");
				}
			//	mGatt.setCharacteristicNotification(,true);
			}
		});
	}

closed 2017-11-26 by Jasonchenlijian