Jasonchenlijian/FastBle #74: 老是提示写入数据请求失败BleException { code=101, description='Initiated Exception Occurred! '}

taiwandao, 2017-05-27:
` private void write(byte[] data) {

        Log.i(TAG,">>>>写入数据");
        bleManager.writeDevice(
                UUID_SERVICE,
                UUID_WRITE,
                data,
                new BleCharacterCallback() {
                    @Override
                    public void onSuccess(BluetoothGattCharacteristic characteristic) {
                        Log.i(TAG, "数据写入成功，响应原始数据：" + AESUtils.BytetohexString(characteristic.getValue()));
                        Log.i(TAG, "数据写入成功，响应内容：" + HexUtil.encodeHexStr(characteristic.getValue()));
                      
                        try {
                            Log.i(TAG, "数据写入成功，解密响应内容：" + AESUtils.decrypt(characteristic.getValue()));
                            Log.i(TAG, "characteristic.getValue()：" + characteristic.getValue());
                        } catch (Exception e) {
                            e.printStackTrace();
                        }

                    }

                    @Override
                    public void onFailure(BleException exception) {
                       
                        Log.i(TAG, "写入数据请求失败" + exception.toString());
                    }
                });
    }`

正常写入数据，一直提示101错误

Jasonchenlijian, 2017-05-28:
void write(byte[] data) 方法写入的时机前后分别做了什么动作？

Jasonchenlijian, 2017-06-29:
写入失败的异常，可以参考Demo或文档中的经验采用合适的方式调用API。

closed 2017-06-29 by Jasonchenlijian