Jasonchenlijian/FastBle #81: writeDevice之后回调的success怎么没有蓝牙设备响应的内容的？要怎么获取啊？

tyousyuka, 2017-06-21:
RT

Jasonchenlijian, 2017-06-21:
write是 app to device.  read, notify, indicate是device to app.

tyousyuka, 2017-06-21:
[image]
像这样吗，但是我执行完writeDevice再执行readDevice，readDevice的onFailure里面直接报错：BleException { code=101, description='Initiated Exception Occurred! '}，求指教

Jasonchenlijian, 2017-06-21:
这个要看你硬件协议咯，到底用到了哪个特征？用的是read还是notify？

tyousyuka, 2017-06-21:
@Jasonchenlijian 换notify也是了，什么返回都没有，求助

Jasonchenlijian, 2017-06-21:
还是要看你的硬件通信协议咯，特征值用对了没有？

tyousyuka, 2017-06-21:
@Jasonchenlijian 特征值不对的话，那writeDevice应该也不会有返回success了吧？

Jasonchenlijian, 2017-06-21:
还是要看你的硬件通信协议，可能你的write是用对了。但device to app，是通过哪个特征值返回？是用的notify还是其他？ 如果是用notify，是不是应该先监听通知，即notify，再进行write？

tyousyuka, 2017-06-21:
@Jasonchenlijian 好的，下午跟硬件的同时咨询一下，谢谢呀

tyousyuka, 2017-06-21:
@Jasonchenlijian  现在用的都是红色里面的，扫出来有这些
[image]

GATT 透传服务以及特征的 UUID如下，应该是怎么样的呀，同事也不懂，郁闷了，帮忙看看哇

[image]

Jasonchenlijian, 2017-06-21:
你的硬件同事应该是最了解的，他应该告诉你device to app 用的是哪个特征值，用的是什么方式。 目前这种情况，你只能去一个个试了。 我建议你先用第三个特征值去notify，再用第四个特征值去write试试吧，祝你好运。

tyousyuka, 2017-06-23:
@Jasonchenlijian onCharacteristicChanged没有回调，是需要什么操作嘛？

Jasonchenlijian, 2017-06-29:
@tyousyuka 需要确认一下 你的外设是否真实地发送数据给APP了？

xwpeng, 2017-08-03:
接不到外设给app发的数据

closed 2017-11-26 by Jasonchenlijian