Jasonchenlijian/FastBle #83: writeDevice() 和notify()都能发送和响应数据,但是每次都有一个异常,这是为什么..

moyihen, 2017-06-21:
I/FastBle: BleGattCallback：onCharacteristicWrite 
06-21 14:36:27.563 17955-17967/com.accuree.yunlock I/Item_Devie_Activity: 4--写入成功02100313
06-21 14:36:27.611 17955-17968/com.accuree.yunlock I/FastBle: BleGattCallback：onCharacteristicChanged 
06-21 14:36:27.612 17955-17968/com.accuree.yunlock I/Item_Devie_Activity: 5--onSuccess通知02060305
06-21 14:36:27.612 17955-17968/com.accuree.yunlock D/BluetoothGatt: cancelOpen() - device: 94:AC:F6:7D:1E:3F
06-21 14:36:27.617 17955-17968/com.accuree.yunlock D/BluetoothGatt: refresh() - device: 94:AC:F6:7D:1E:3F
06-21 14:36:27.619 17955-17968/com.accuree.yunlock I/FastBle: refreshDeviceCache, is success:  true
06-21 14:36:27.619 17955-17968/com.accuree.yunlock D/BluetoothGatt: close()
06-21 14:36:27.619 17955-17968/com.accuree.yunlock D/BluetoothGatt: unregisterApp() - mClientIf=5
06-21 14:36:27.623 17955-17968/com.accuree.yunlock W/BluetoothGatt: Unhandled exception in callback
                                                                    java.util.ConcurrentModificationException
                                                                        at java.util.HashMap$HashIterator.nextEntry(HashMap.java:851)
                                                                        at java.util.HashMap$EntryIterator.next(HashMap.java:891)
                                                                        at java.util.HashMap$EntryIterator.next(HashMap.java:890)
                                                                        at com.clj.fastble.bluetooth.BleBluetooth$4.onCharacteristicChanged(BleBluetooth.java:451)
                                                                        at android.bluetooth.BluetoothGatt$1.onNotify(BluetoothGatt.java:364)
                                                                        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:177)
                                                                        at android.os.Binder.execTransact(Binder.java:565)
06-21 14:36:27.624 17955-17968/com.accuree.yunlock D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=94:AC:F6:7D:1E:3F
06-21 14:36:27.624 17955-17968/com.accuree.yunlock W/BluetoothGatt: Unhandled exception in callback
                                                                    java.lang.NullPointerException: Attempt to invoke virtual method 'void android.bluetooth.BluetoothGattCallback.onConnectionStateChange(android.bluetooth.BluetoothGatt, int, int)' on a null object reference
                                                                        at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:191)
                                                                        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:70)
                                                                        at android.os.Binder.execTransact(Binder.java:565)

tyousyuka, 2017-06-22:
你成功了没

moyihen, 2017-06-22:
@tyousyuka 你说的成功啥意思...我的能正常发送 和接收消息  就是会有一个上面的异常.

tyousyuka, 2017-06-22:
先notify？再write？再read？特征值大概怎么设置的呢？

moyihen, 2017-06-22:
我是连接成功 notify write 没有用到read  .特征值你得和你通信的那一方协商好啊 什么是notify write....

moyihen, 2017-06-22:
@tyousyuka

tyousyuka, 2017-06-23:
@Jasonchenlijian    onCharacteristicChanged回调有什么特殊设置嘛？我这边notify write 一只么反应

Jasonchenlijian, 2017-06-29:
@tyousyuka 假如notify成功之后就可以等待外设向APP发送数据了，并没有特殊设置。