Jasonchenlijian/FastBle #91: 有个问题不理解求指教

fujingxie, 2017-07-04:
一个简单的使用场景：打开蓝牙，在主线程扫描设备，连接，连接成功并发现服务之后，在onServicesDiscovered的异步回调方法中，延时100ms，再切换到主线程，再去调用notify、write等方法。这就是一个基本的操作。
意思是不是切换到主线程再去注册notify的监听 和wirte的操作。因为我有个场景是 在onServicesDiscovered的回调中直接setCharacteristicNotification和write同时进行，但是只有write成功的回调 ,notify却一直没有回调。请指教一下，谢谢了

Jasonchenlijian, 2017-07-05:
通知和写操作不要通知进行，中间间隔一段时间，并且都在主线程，试试看。

fujingxie, 2017-07-05:
@Jasonchenlijian  请问中间间隔一段时间是怎么处理的  
//todo 通知
new Handler().postDelayed(new Runnable(){   

    public void run() {   

    //execute the task   

    }   

 }, delay);   
//todo 写
还有通知和写的BluetoothGatt  取值onServicesDiscovered(BluetoothGatt gatt, int status)的gatt有没有问题？谢谢大神指导下

Jasonchenlijian, 2017-07-05:
通知和写，不都有单独的方法吗，没有必要使用回调方法里面的gatt。

closed 2017-11-28 by Jasonchenlijian