Jasonchenlijian/FastBle #93: 我现在发现一个读写问题

Xanthuim, 2017-07-07:
我有个蓝牙芯片，使用这个去连接然后写入2137，正常会返回一串数据。但是现在我发现当我不调试会抛出BleException的102异常信息，而我调试一步步的走则能得到这串数据。这是什么鬼？

Xanthuim, 2017-07-07:
跟踪后发现是这里出错了：
在BleConnector类的writeCharacteristic方法中
return handleAfterInitialed(getBluetoothGatt().readCharacteristic(getCharacteristic()),
                    bleCallback);
getBluetoothGatt().readCharacteristic(getCharacteristic())返回的是false，看到日志信息是：
 readCharacteristic: mDeviceBusy = true, and return false
不知道为啥。我调试没问题，正常运行就有错，难道是调试时执行的慢可以得到，正常就繁忙？没道理啊，我点的也不快。

Jasonchenlijian, 2017-07-07:
onInitiatedResult(boolean result) 返回什么值？

Xanthuim, 2017-07-07:
知道原因了，readCharacteristic这个必须在onCharacteristicRead方法执行了才返回true的，而你这个值同步方法，所有直接运行时false（原因是onCharacteristicRead还没有被调用）。这个地方算是一个bug。

Xanthuim, 2017-07-07:
再次跟踪，发现注释掉这个方法的调用就可以成功获取：
setCharacteristicNotification
我不知道这个设置通知的作用是什么。

Heyxiang, 2017-07-26:
我几天碰到楼上问题，读设备名称，版本没有问题，但是当读取电池电量时候，就读不出来了，跟踪发现
    /**
     * read
     */
    public boolean readCharacteristic(BleCharacterCallback bleCallback, String uuid_read) {
        if (getCharacteristic() != null
                && (characteristic.getProperties() & BluetoothGattCharacteristic.PROPERTY_READ) > 0) {

            BleLog.d(TAG, getCharacteristic().getUuid()
                    + "\ncharacteristic.getProperties(): " + getCharacteristic().getProperties()
                    + "\ncharacteristic.getValue(): " + Arrays.toString(getCharacteristic().getValue()));

           // setCharacteristicNotification(getBluetoothGatt(), getCharacteristic(), false, bleCallback);
            handleCharacteristicReadCallback(bleCallback, uuid_read);
            return handleAfterInitialed(getBluetoothGatt().readCharacteristic(getCharacteristic()), bleCallback);

        } else {
            if (bleCallback != null) {
                bleCallback.onFailure(new OtherException("this characteristic not support read!"));
                bleCallback.onInitiatedResult(false);
            }
            return false;
        }
    }
我中间注释的那行代码 setCharacteristicNotification(getBluetoothGatt(), getCharacteristic(), false, bleCallback); 出现问题，在进行读的时候去去打开通知，执行了write操作，导致读失败，解决方法，把那行代码注释就可以了。

closed 2017-11-26 by Jasonchenlijian