Jasonchenlijian/FastBle #95: 发服务成功的回调会有两次

TearBack, 2017-07-12:
我一直在使用你的这个项目。但是总是发现有一些小毛病。上一个版本1.1.1使用的时候，就发现，连接成功onConnectSuccess的回调会回调两次。这次1.2.1连接成功没有回调两次了，但是服务发现onServicesDiscovered又回调了两次。看了一下日志，确实是有两次discoverServices调用，而我在连接成功，只调用了一次发现服务。

I/FastBle: BleGattCallback：onConnectSuccess 
I/System.out: tsdabby：连接成功，开始发现服务
D/BluetoothDevice: mAddress: 00:07:80:D1:8B:B6
D/BluetoothGatt: discoverServices() - device: 00:07:80:D1:8B:B6
D/BluetoothDevice: mAddress: 00:07:80:D1:8B:B6
D/BluetoothDevice: mAddress: 00:07:80:D1:8B:B6
D/BluetoothGatt: discoverServices() - device: 00:07:80:D1:8B:B6
D/BluetoothDevice: mAddress: 00:07:80:D1:8B:B6
/BluetoothGatt: onGetService() - Device=00:07:80:D1:8B:B6 UUID=00001800-0000-1000-8000-00805f9b34fb
/BluetoothDevice: mAddress: 00:07:80:D1:8B:B6
/BluetoothGatt: onGetService() - Device=00:07:80:D1:8B:B6 UUID=0000180a-0000-1000-8000-00805f9b34fb
/BluetoothDevice: mAddress: 00:07:80:D1:8B:B6

@Override
            public void onConnectSuccess(BluetoothGatt gatt, int status) {
                //连接成功的回调
                LogUtil.sout("连接成功，开始发现服务");
                gatt.discoverServices();
            }

Jasonchenlijian, 2017-07-12:
你好，1.2.x版本上，在onConnectSuccess回调方法中，不需要再手动调用gatt.discoverServices()方法了。这一点可以在示例代码中看到。抱歉针对这一点，对之前使用1.1.x老版本的用户没有着重说明。

TearBack, 2017-07-12:
真是太感谢了，这样就完美了

closed 2017-07-12 by TearBack