Jasonchenlijian/FastBle #105: lenovo android 7.0 7.1有问题

gjjofsfe, 2017-07-27:
品牌lenovo ，android 7 or 7.1.1扫描不到蓝牙设备。
同时使用了bletool 和 nRF MasterControlPanel 都能扫描且连接成功。

wuxjim, 2017-07-30:
遇到同样的问题。在华为荣耀平板 2 上也是不能用。。应该是跟 andriod 7.0 有关系

gjjofsfe, 2017-07-31:
使用老版本的startscan和5.0以上的scanner，都是扫描不到任何设备，还是找不到原因...

gjjofsfe, 2017-07-31:
但是不知道为什么bletool 和 nRF MasterControlPanel 没有开定位服务也能扫描且连接成功呢。。。

yishangfei, 2017-09-26:
蓝牙连接怎么还要定位呀

MisterBowie, 2017-10-27:
我也遇到 nrf 能找到服务  框架就不能,已经添加
 <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
两个权限.  开启定位服务什么意思 各位

MisterBowie, 2017-10-27:
android.permission.ACCESS_COARSE_LOCATION" />
android.permission.ACCESS_FINE_LOCATION" />

Jasonchenlijian, 2017-10-27:
各位注意两点：
1. 6.0以上的系统 扫描蓝牙外设需要额外获取那两个位置权限，这是必须。
2. 部分6.0以上的系统的手机，可能还需要手动打开位置信息，可以在APP里面引导用户去设置界面手动打开。

MisterBowie, 2017-10-27:
找到原因了  和楼主说的一样 虽然我申明了两个额外的定位权限 但是没有让用户去打开,导致无法搜索蓝牙.

closed 2017-10-27 by Jasonchenlijian

TYUpya, 2020-01-02:
我现在也遇到了同样的问题。
我的 魅族手机（型号:M711C、系统7.0）必须要 用户开启位置信息才能扫描到蓝牙设备，
我的 华为手机（型号:PRA-AL00X、系统8.0.0）就 不需要 用户手动开启位置信息就能扫描到蓝牙设备
所以并不是所有6.0以上手机都得开启位置信息(gps)
我试用了一款名为（nRF Connect）的蓝牙通讯软件，它就能准确识别哪些设备需要用户开启位置信息(gps)或不需要，跟手机系统版本无关。
请问（nRF Connect）是怎么实现的呢？
@gjjofsfe @wuxjim @yishangfei @MisterBowie @Jasonchenlijian

MisterBowie, 2020-01-02:
> 我现在也遇到了同样的问题。
> 我的 魅族手机（型号:M711C、系统7.0）必须要 用户开启位置信息才能扫描到蓝牙设备，
> 我的 华为手机（型号:PRA-AL00X、系统8.0.0）就 不需要 用户手动开启位置信息就能扫描到蓝牙设备
> 所以并不是所有6.0以上手机都得开启位置信息(gps)
> 我试用了一款名为（nRF Connect）的蓝牙通讯软件，它就能准确识别哪些设备需要用户开启位置信息(gps)或不需要，跟手机系统版本无关。
> 请问（nRF Connect）是怎么实现的呢？
> @gjjofsfe @wuxjim @yishangfei @MisterBowie @Jasonchenlijian

对于我们一般开发者 一些手机需要打开位置服务才能搜索到蓝鞋     为了适配最好一视同仁打开