Jasonchenlijian/FastBle #116: 你好。我并没有在BleGattCallback回调里面找到onServicesDiscovered()方法 请问是不是版本的问题

LinkSy, 2017-08-25:
同上

MisterBowie, 2017-10-26:
我也是

MisterBowie, 2017-10-26:
我也是

Jasonchenlijian, 2017-10-26:
onServicesDiscovered()并不是强制必须复写的方法。如果需要，请手动复写。

MisterBowie, 2017-10-26:
也没法重写,
对应的GATTcallback类
package com.clj.fastble.conn;

import android.bluetooth.BluetoothGatt;
import android.bluetooth.BluetoothGattCallback;
import com.clj.fastble.data.ScanResult;
import com.clj.fastble.exception.BleException;

public abstract class BleGattCallback extends BluetoothGattCallback {
    public BleGattCallback() {
    }

    public void onFoundDevice(ScanResult scanResult) {
    }

    public void onConnecting(BluetoothGatt gatt, int status) {
    }

    public abstract void onConnectError(BleException var1);

    public abstract void onConnectSuccess(BluetoothGatt var1, int var2);

    public abstract void onDisConnected(BluetoothGatt var1, int var2, BleException var3);
}

是导包为题吗?

MisterBowie, 2017-10-26:
版本com.clj.fastble:FastBleLib:1.2.1

Jasonchenlijian, 2017-10-26:
onServicesDiscovered() 是BluetoothGattCallback的方法。

MisterBowie, 2017-10-26:
不好意思 没看见,我也看了BluetoothGattCallback类 过的快了没看见 studio复写的时候只看了BleGattCallback的能复写的方法  sorry

closed 2017-10-27 by Jasonchenlijian