Jasonchenlijian/FastBle #117: 当发送数据长度大于20时，建议加上以下方法，请采纳：

carozhu, 2017-08-25:
ServerBufferSize = 20；
if (dataLen > ServerBufferSize) {
      int count = 0;
            int offset = 0;
            while (offset < dataLen) {
                if ((dataLen - offset) < ServerBufferSize)
                    count = dataLen - offset;
                else
                    count = ServerBufferSize;
                byte tempArray[] = new byte[count];
                System.arraycopy(bs, offset, tempArray, 0, count);
                ret = bleManager.writeDevice(
                        UUID_SERVICE,
                        UUID_WRITE,
                        tempArray,
                        new BleCharacterCallback() {
                            @Override
                            public void onSuccess(BluetoothGattCharacteristic characteristic) {
                                Log.i(TAG, "write opendoor request onSuccess------------");
                            }

                            @Override
                            public void onFailure(BleException exception) {
                                Log.i(TAG, "write opendoor request onFailure------------");
                            }

                            @Override
                            public void onInitiatedResult(boolean result) {
                                Log.i(TAG, "write opendoor request onInitiatedResult----");
                            }
                        });
                if (!ret)
                    return ret;
                offset = offset + count;
                SystemClock.sleep(10);
            }
}

AndroidBirdBoom, 2017-09-21:
请问，我使用您的方法做分包处理了，写入的时候失败率比较高，请问您这边怎么处理分包问题的，谢谢

NoddyLiu, 2017-12-22:
有效，谢谢您

closed 2018-01-08 by Jasonchenlijian

longzekai, 2019-11-23:
其实楼主的也可以，只不过框架默认是拆包后间隔0ms就发送下一条数据，而有些设备，不能这么快就写入数据，所以，只要修改下发送时间间隔就没问题了。