Jasonchenlijian/FastBle #128: 要是一个Character又能notify，又能write，你只能存在一个回调

kanshenmekan, 2017-09-28:
要是一个Character又能notify，又能write，你只能存在一个回调。。怎么弄 @Jasonchenlijian

santa-cat, 2017-11-10:
我也发现了这个问题，我现在做的只能把一个回调传null @Jasonchenlijian

kid2682, 2017-11-24:
我也碰到这个问题，修改了一下源码。临时解决了一下。
BleManager.java
`public boolean notify(String uuid_service,
                          String uuid_notify,
                          BleCharacterCallback callback) {
        return bleBluetooth.newBleConnector()
                .withUUIDString(uuid_service, uuid_notify, null)
                .enableCharacteristicNotify(callback, "notify_"+uuid_notify);
    }`

BleConnector.java
`	private void handleCharacteristicNotificationCallback(final BleCharacterCallback bleCallback,
	                                                      final String uuid_notify) {
		if (bleCallback != null) {

			listenAndTimer(bleCallback, MSG_NOTIFY_CHA, uuid_notify, new BluetoothGattCallback() {
				AtomicBoolean msgRemoved = new AtomicBoolean(false);

				@Override
				public void onCharacteristicChanged(BluetoothGatt gatt,
				                                    BluetoothGattCharacteristic characteristic) {

					if (!msgRemoved.getAndSet(true)) {
						handler.removeMessages(MSG_NOTIFY_CHA, this);
					}
					if (characteristic.getUuid().equals(UUID.fromString(uuid_notify.replace("notify_","")))) {
						bleCallback.onSuccess(characteristic);
					}
				}
			});
		}
	}`

Jasonchenlijian, 2017-11-26:
是的，你提的问题很好。楼上提供的解决方法也是正确的。之前的版本一直没有太多关注这一点，如果一个特性有两种操作，当对其进行第二种操作的时候，就会覆盖之前那个操作的数据。
解决方法是：

如果使用的是2.1.0版本之前的代码，可以修改源码中的回调Hashmap的key值，原先的key值仅仅是这个特性的uuid，可以用uuid+操作属性替代。
2.1.0版本之后已经修复这个问题。

closed 2017-11-26 by Jasonchenlijian