Jasonchenlijian/FastBle #131: android 6.0设备出现的问题

youxikaifa, 2017-10-12:
在android4.4的测试机上运行正常，在6.0上运行会扫描不到设备，ok，我把TargetSdk降到21，现在可以扫描到了，也能连接上外设，但是在传输数据的时候回一直断掉，设备会自动关机，在4.4的设备上偶尔会出现这种情况，请问一下大概是什么原因造成的。

Jasonchenlijian, 2017-10-12:
6.0的问题在于Dangerous Permissions，详细看项目主页上的说明。
断开的问题最好请检查设备。

youxikaifa, 2017-10-14:
连接设备后，未调用onServiceDiscovered回调，onConnectionStateChange回调连接成功，这是怎么回事

Jasonchenlijian, 2017-10-18:
onServiceDiscovered是ServiceDiscovered，onConnectionStateChange是连接状态改变。顺序是：连接成功之后，检索服务。

youxikaifa, 2017-10-18:
现在有个问题想请教下，我发送数据的时候，没有返回数据给我，onCharacteristicChanged也没有调用，设置了通知，通道也有通知属性，请问是怎么回事

Jasonchenlijian, 2017-10-18:
请给出流程代码参考

youxikaifa, 2017-10-18:
在确定蓝牙连接成功之后，一次调用下面两个方法，
public void enable_channel(){
        if (mBluetoothGatt == null) {
            return;
        }
        BluetoothGattService service = mBluetoothGatt.getService(SampleGattAttributes.USER_SERVICE_UUID);
        BluetoothGattCharacteristic characteristic;
        if (service != null) {
            characteristic = service.getCharacteristic(SampleGattAttributes.READ_WRITE_NOTIFY_UUID);
        } else {
            return;
        }
        boolean set = mBluetoothGatt.setCharacteristicNotification(characteristic, true);
        Log.d(TAG, " setnotification = " + set);

        for(BluetoothGattDescriptor dp:characteristic.getDescriptors()){
            dp.setValue(BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE);
            boolean success = mBluetoothGatt.writeDescriptor(dp);
            Log.d(TAG, "writing enable descriptor:" + success);
        }
    }

=================================

public void sendInstruction(){

        byte[] sendData = {(byte) 0xAA, (byte) 0xBE, 0x08, 0x03, (byte) 0xFA, 0x36, (byte) 0xD3, 0x31, (byte) 0xBE, 0x40, 0x27, (byte) 0xE6};

        BluetoothGattService service = mBluetoothGatt.getService(SampleGattAttributes.USER_SERVICE_UUID);
        if (service == null)
            return;
        BluetoothGattCharacteristic gg = service.getCharacteristic(SampleGattAttributes.READ_WRITE_NOTIFY_UUID);

        if (gg == null) {
            return;
        }
//        gg.setValue(build_cmd(sendData));
        gg.setValue((new EncodeUtil()).encodeMessage(build_cmd(sendData)));
        mBluetoothGatt.writeCharacteristic(gg);
    }

Jasonchenlijian, 2017-10-29:
先设置通知，再进行write，关注一下是否成功write。另外，进行write之前最好延迟一点时间。

closed 2017-11-06 by Jasonchenlijian