Jasonchenlijian/FastBle #133: writeDevice之后没有任何的回调被触发，没有任何的反应，特征码是正确的

zj2050, 2017-10-14:
return bleManager.writeDevice(
                UUID_SERVICE,
                UUID_WRITE,
                HexUtil.hexStringToBytes(SAMPLE_WRITE_DATA),
                new BleCharacterCallback() {
                    @Override //数据传输回调
                    public void onSuccess(BluetoothGattCharacteristic characteristic) {
                        Looper.prepare();
                        Toast.makeText(mContext,String.valueOf(HexUtil.encodeHex(characteristic.getValue())),Toast.LENGTH_LONG).show();
                        img_loading.clearAnimation();
                    }

                    @Override
                    public void onFailure(BleException exception) {
                        img_loading.clearAnimation();
                    }

                    @Override  //操作成功与否的回调；
                    public void onInitiatedResult(boolean result) {

                        img_loading.clearAnimation();
                    }

                });

以上任何一个断点都到不了
 HexUtil.hexStringToBytes(SAMPLE_WRITE_DATA),
断点能到

我是在

                    @Override
                    public void onServicesDiscovered(BluetoothGatt gatt, int status) {
                        globalConfig.Delay(1); //延时是为等待蓝牙设备准备好

                        write();
                    }

里面调用 写方法的，谢谢

zj2050, 2017-10-14:
4.3 和 7.0 的都不行，手机是 三星 noteii 和 s7

zj2050, 2017-10-14:
设备端要求发送的 是 4 个 0xaa， 我传进去的数据是 SAMPLE_WRITE_DATA = "aaaaaaaa";     
这个有问题吗？

Qingtian239113, 2017-10-16:
可以尝试
for (BluetoothGattService service : gatt.getServices()) {
                    if (service.getUuid().toString().equalsIgnoreCase("11223344-5566-7788-99AA-BBCCDDEEFF00")) {
                        BluetoothGattCharacteristic characteristic =
                                service.getCharacteristic(UUID.fromString("00004a5b-0000-1000-8000-00805f9b34fb"));
                        if (characteristic != null) {
                            characteristic.setValue(writeBytes);
                            gatt.writeCharacteristic(characteristic);
                            gatt.setCharacteristicNotification(characteristic, true);
                        }
                    }
                }
这样去写入输入

zj2050, 2017-10-17:
@Qingtian239113 谢谢，直接用service 那个方式去调用写好像可以， 但自己直接blemanager 的时候没反应，不知道为啥

closed 2017-10-17 by zj2050

ervinsae, 2017-12-11:
同样出现write成功后不走任何回调

Jasonchenlijian, 2017-12-11:
标注一下使用的版本，我了解情况后给你解答。

ervinsae, 2017-12-11:
使用的是2.1.0版本
 BleManager.getInstance().write(device, UUIDs.serviceUUID, UUIDs.writeUUID, data, callback);

callback中的回调都没有调用，数据确定已经发出去了

ervinsae, 2017-12-11:
BleHelper.getInstance().sendData(device, BleHelper.COMMAND_START, new BleWriteCallback() {
                        @Override
                        public void onWriteSuccess() {
                            Log.d("开始success","成功");

                        }

                        @Override
                        public void onWriteFailure(BleException exception) {
                            Log.d("开始failure",exception.getDescription());
                        }
                    }

Jasonchenlijian, 2017-12-11:
2.1.0版本有bug，用之前的版本（2.0.1）或者之后的版本（2.1.1，建议最新的2.1.5），我已经在文档中去除了这个版本。
另外，发送数据建议在主线程中进行。