Jasonchenlijian/FastBle #144: 有一个问题想请问一下各位大佬，求解答~~

KingOfSu, 2017-11-14:
就是我连接了蓝牙之后，  使用了     bleManager.closeBluetoothGatt();    蓝牙断开了，但是当我重新连接的时候就无法连接上去了，请问需要如何才能重新连接呢！求大佬们帮我一下拉！么么哒

jhxiaoming, 2017-11-14:
说明里面不是有吗？
连接断开之后的重连很简单，在void onDisConnected(BluetoothGatt gatt, int status, BleException exception)调用boolean gatt.connect()方法即可，当外设再次处于可连接状态时，就会自动连上。
能不能认真看下说明？

closed 2017-11-26 by Jasonchenlijian

douya40, 2020-07-04:
实践过不行

douya40, 2020-07-04:
@Override
            public void onDisConnected(boolean isActiveDisConnected, BleDevice bleDevice, BluetoothGatt gatt, int status) {
                Log.i("zhaocheng","onDisConnected");
                gatt.connect();
                if (isActiveDisConnected) {
//                    Toast.makeText(MainActivity.this, getString(R.string.active_disconnected), Toast.LENGTH_LONG).show();
                } else {
//                    Toast.makeText(MainActivity.this, getString(R.string.disconnected), Toast.LENGTH_LONG).show();
//                    ObserverManager.getInstance().notifyObserver(bleDevice);
                }
            }