Jasonchenlijian/FastBle #147: onServicesDiscovered回调后调用notify失败

lm11000, 2017-11-21:
我在BleGattCallback里的onServicesDiscovered()后调用notify,检查了uuid无误.但是会弹出log
W/BleConnector: characteristic.getProperties():16
然后便走向了回调onInitiatedResult().其result偶尔为true偶尔为false.
迫切希望作者大大解答,谢谢

lm11000, 2017-11-21:
已解决.我原本在onInitiatedResult().回调中写了disconnect方法.因为我看范例中这里是失败的回调.
而事实上第一次notify的时候会走这里,接下来便走了onSuccessful()回调.
所以我去掉了disconnect的代码就迎刃而解了

lm11000, 2017-11-21:
但是同时我有另一个问题.
我在onDisconnected()的回调中写了scanAndConnect()方法.在设备断电的情况下是可以重连的.
但是在我调用closeBluetoothGatt()方法以后.我发现并不会走onDisconnect()或者onConnectError()的回调.
如果我想监听我手动断开连接的画,应该监听哪一个回调呢?
或者除了closeBluetoothGatt()方法之外还有别的断开连接的方法吗?

Jasonchenlijian, 2017-11-21:
调用closeBluetoothGatt()之后，之前建立的回调都会被清空，所以onDisconnect()或者onConnectError()都不会起作用了。我建议，如果是程序手动调用closeBluetoothGatt()，此时此刻本身就表示断开连接，不需要回调来辅助判断。建议在closeBluetoothGatt()之后，延迟一段时间后进行其他操作，如重新扫描。

lm11000, 2017-11-21:
好的,谢谢!

closed 2017-11-26 by Jasonchenlijian