Jasonchenlijian/FastBle #149: android7.0扫描不到设备

suhuangqiang, 2017-11-24:
android7.0扫描不到任何设备，5.1的扫描也没有看到首届设备

Jasonchenlijian, 2017-11-26:
抱歉，我并不能准确定位你问题的原因。
可能原因：
1. 这个框架只能扫描出外围设备，而不能是手机。
2. 这个框架智能扫描出BLE设备，而非经典蓝牙模块。
3. 是否注册了蓝牙权限和模糊位置权限？部分手机上还需要打开定位功能。 
4. 2.0.0和2.0.1版本上，存在一个bug，假如该BLE模块没有广播名，则不会被扫描出来，建议不要使用这两个版本。

suhuangqiang, 2017-11-27:
之前是用2.0.1的，现在换成2.1.0的了
有个问题，就是我用华P1的还是扫描不到，用一加的扫描到了，都Android7以上的系统
然后用6.0一下的系统还有名字，7.0以上的名字就都为空了

xiaoniu, 2017-11-30:
因为搜索蓝牙会暴露手机的位置信息，所以在Android6.0以后需要申请位置权限才能搜索到蓝牙设备。我的一加5，Android版本7.1.1，之前也是搜索不到蓝牙设备，但是把位置服务开启后就可以了。

DCZLQR, 2017-12-07:
我自己用的6.0的手机可以正常扫描，4.4的也可以，公司5.0的也不行，请问解决了吗，版本是2.1.4，之前的版本是1.几的，都可以正常扫描

TYUpya, 2020-01-02:
我现在也遇到了同样的问题。
我的 魅族手机（型号:M711C、系统7.0）必须要 用户开启位置信息才能扫描到蓝牙设备，
我的 华为手机（型号:PRA-AL00X、系统8.0.0）就 不需要 用户手动开启位置信息就能扫描到蓝牙设备
所以并不是所有6.0以上手机都得开启位置信息(gps)
我试用了一款名为（nRF Connect）的蓝牙通讯软件，它就能准确识别哪些设备需要用户开启位置信息(gps)或不需要，跟手机系统版本无关。
请问（nRF Connect）是怎么实现的呢？
@suhuangqiang @Jasonchenlijian @xiaoniu @DCZLQR

closed 2021-12-07 by Jasonchenlijian