Jasonchenlijian/FastBle #153: 多次onScanComplete()回调中再次调用scan会出现不执行onScanComplete()回调

lm11000, 2017-11-27:
如题.我本来是想做不停止扫描,如果设备没有搜到,就要一直保持扫描,但是现在出现了一个问题就是onScanComplete()第一次可以再scan(),这一次scan()就不会执行onScanComplete()回调了

Jasonchenlijian, 2017-11-27:
private void startScan() {
        BleManager.getInstance().scan(new BleScanCallback() {
            @Override
            public void onScanStarted(boolean success) {

            }

            @Override
            public void onScanning(BleDevice bleDevice) {

            }

            @Override
            public void onScanFinished(List<BleDevice> scanResultList) {
                startScan();
            }
        });
    }

经过测试是可以完成你的需求的。我贴的是2.1.0版本的调用形式，可能与你的版本不同，但逻辑是相同的。

lm11000, 2017-11-27:
谢谢!我定位了真正问题.我插入了第二个符合规则的蓝牙之后,APP能够马上扫描到设备,代表扫描一直在运行,没有停止,故而没有走入complete()...而原本我设定的扫描周期是8秒.
不过就目前而言不停止扫描也能符合我的需求,但是这个问题我还是给你提出来.

Jasonchenlijian, 2017-11-27:
谢谢你的验证。这个必须符合规则才会回调complete()的缺陷我也发现了，不太合理，在2.1.0版本中已经修复这个问题，不论扫描结果如何最后都会回调onScanFinished()，你可以一试。

lm11000, 2017-11-28:
2.1更新挺大的啊..然而,事实上这个问题并没有修复.
依然是在complete()后--scan(),扫描将不会停止
11-28 11:08:38.423 11608-11608/cn.askj.ebike I/MainActivity: onScanStarted: 扫描开启
11-28 11:08:38.618 11608-11608/cn.askj.ebike I/FastBle: onScanning  ------  name: EBIKE  mac: 09:01:00:47:AD:58  Rssi: -40  scanRecord: 0201060aff68820109010047ad58030201f007094542494b4500000000000000000000000000000000000000000000000000000000000000000000000000
11-28 11:08:38.678 11608-11608/cn.askj.ebike I/MainActivity: onScanning: scanResultMacEBIKE
11-28 11:08:38.678 11608-11608/cn.askj.ebike I/MainActivity: onScanning: 扫描中
11-28 11:08:38.679 11608-11608/cn.askj.ebike I/FastBle: onScanning  ------  name: EBIKE  mac: 09:01:00:E8:59:9C  Rssi: -52  scanRecord: 0201060aff688201090100e8599c030201f007094542494b4500000000000000000000000000000000000000000000000000000000000000000000000000
11-28 11:08:38.694 11608-11608/cn.askj.ebike I/MainActivity: onScanning: scanResultMacEBIKE
11-28 11:08:38.694 11608-11608/cn.askj.ebike I/MainActivity: onScanning: 扫描中
11-28 11:08:46.448 11608-11608/cn.askj.ebike I/MainActivity: onScanComplete: 扫描结束
11-28 11:08:46.485 11608-11608/cn.askj.ebike I/MainActivity: onScanStarted: 扫描开启
11-28 11:08:46.554 11608-11608/cn.askj.ebike I/FastBle: onScanning  ------  name: EBIKE  mac: 09:01:00:E8:59:9C  Rssi: -54  scanRecord: 0201060aff688201090100e8599c030201f007094542494b4500000000000000000000000000000000000000000000000000000000000000000000000000
11-28 11:08:46.587 11608-11608/cn.askj.ebike I/MainActivity: onScanning: scanResultMacEBIKE
11-28 11:08:46.587 11608-11608/cn.askj.ebike I/MainActivity: onScanning: 扫描中
11-28 11:08:46.589 11608-11608/cn.askj.ebike I/FastBle: onScanning  ------  name: EBIKE  mac: 09:01:00:47:AD:58  Rssi: -39  scanRecord: 0201060aff68820109010047ad58030201f007094542494b4500000000000000000000000000000000000000000000000000000000000000000000000000
11-28 11:08:46.607 11608-11608/cn.askj.ebike I/MainActivity: onScanning: scanResultMacEBIKE
11-28 11:08:46.607 11608-11608/cn.askj.ebike I/MainActivity: onScanning: 扫描中

Jasonchenlijian, 2017-11-28:
我不太理解你的意思，“在complete()后--scan(),扫描将不会停止”，前一次扫描结束以后，重新调用scan，当然会继续扫描啊？

lm11000, 2017-11-28:
重新调用scan()继续扫描肯定是应该的,但是之前我设置的扫描规则的setScanTimeOut(8000)却失效了.

Jasonchenlijian, 2017-11-28:
你的意思是，首先设置了setScanTimeOut(8000)。第一次开启扫描，8秒后调用onScanFinished；然后在onScanFinished方法中再次调用scan方法，不会回调onScanFinished方法了吗？或者说，不是8秒以后调用？

lm11000, 2017-11-28:
首先设置了setScanTimeOut(8000)。第一次开启扫描，8秒后调用onScanFinished；然后在onScanFinished方法中再次调用scan方法.
扫描会一直进行,将不会停止.自然就不会回调onScanFinished()了.

Jasonchenlijian, 2017-11-28:
可以在onScanFinished()方法中延迟一下100ms再开启Scan。

lm11000, 2017-11-28:
可行,并且现在挺稳定,没有发现什么BUG.感谢!

closed 2017-11-28 by Jasonchenlijian