Jasonchenlijian/FastBle #155: 扫描蓝牙提示缺少UUID

lovelyYakir, 2017-11-30:
11-30 12:02:47.203 25051-25051/com.awsome.yj.bluetoothdemo E/AndroidRuntime: FATAL EXCEPTION: main
                                                                             Process: com.awsome.yj.bluetoothdemo, PID: 25051
                                                                             java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.UUID[] com.clj.fastble.scan.BleScanRuleConfig.getServiceUuids()' on a null object reference
                                                                                 at com.clj.fastble.BleManager.scan(BleManager.java:206)
                                                                                 at com.awsome.yj.bluetoothdemo.MainActivity$onCreate$1.onClick(MainActivity.kt:47)
                                                                                 at android.view.View.performClick(View.java:5688)
                                                                                 at android.view.View$PerformClick.run(View.java:22407)
                                                                                 at android.os.Handler.handleCallback(Handler.java:751)
                                                                                 at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                 at android.os.Looper.loop(Looper.java:154)
                                                                                 at android.app.ActivityThread.main(ActivityThread.java:6268)
                                                                                 at java.lang.reflect.Method.invoke(Native Method)
                                                                                 at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:911)
                                                                                 at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:801)

以下是调用的方法。已经在Application初始化了。
 if (!BleManager.getInstance().isSupportBle) {
                Toast.makeText(this, "不支持蓝牙!", Toast.LENGTH_SHORT).show();
            } else {
                //支持蓝牙，打开蓝牙
                if (!BleManager.getInstance().isBlueEnable) {
                    BleManager.getInstance()
                            .enableBluetooth()
                }
                BleManager.getInstance().scan(object :BleScanCallback(){
                    override fun onScanning(result: BleDevice?) {

                    }

                    override fun onScanStarted(success: Boolean) {

                    }

                    override fun onScanFinished(scanResultList: MutableList<BleDevice>?) {

                    }
                })

Jasonchenlijian, 2017-11-30:
扫描之前，先配置一下扫描规则：
BleManager.getInstance().initScanRule(scanRuleConfig)

Jasonchenlijian, 2017-11-30:
在2.1.2版本及之前，必须先配置过滤规则再扫描；在2.1.3版本之后可以无需配置，开启默认过滤规则的扫描。

closed 2017-11-30 by Jasonchenlijian