Jasonchenlijian/FastBle #160: BleManager.stopNotify 空指针问题

GoAheadComeOn, 2017-12-06:
没有复现这个bug，但是在bugly统计的日志是有关于这个bug的信息，仔细查看后发现跟特定机型有关机型信息：三星 GT-N7100 | Android 6.0.1,level 23
日志信息：

1 io.reactivex.exceptions.OnErrorNotImplementedException:Attempt to invoke virtual method 'com.clj.fastble.conn.BleConnector com.clj.fastble.bluetooth.BleBluetooth.newBleConnector()' on a null object reference
--
2 io.reactivex.internal.functions.Functions$14.accept(Functions.java:229)
3 ......
4 Caused by:
5 java.lang.NullPointerException:Attempt to invoke virtual method 'com.clj.fastble.conn.BleConnector com.clj.fastble.bluetooth.BleBluetooth.newBleConnector()' on a null object reference
6 com.clj.fastble.BleManager.stopNotify(BleManager.java:218)
7 com.zhiji.work.presenter.vehiclecontrol.ControlDoorPresenter$9.accept(ControlDoorPresenter.java:727)
8 com.zhiji.work.presenter.vehiclecontrol.ControlDoorPresenter$9.accept(ControlDoorPresenter.java:724)
9 io.reactivex.internal.observers.LambdaObserver.onNext(LambdaObserver.java:59)
10 io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.drainNormal(ObservableObserveOn.java:200)
11 io.reactivex.internal.operators.observable.ObservableObserveOn$ObserveOnObserver.run(ObservableObserveOn.java:252)
12 io.reactivex.android.schedulers.HandlerScheduler$ScheduledRunnable.run(HandlerScheduler.java:109)
13 android.os.Handler.handleCallback(Handler.java:739)
14 android.os.Handler.dispatchMessage(Handler.java:95)
15 android.os.Looper.loop(Looper.java:148)
16 android.app.ActivityThread.main(ActivityThread.java:5417)
17 java.lang.reflect.Method.invoke(Native Method)
18 com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:726)
19 com.android.internal.os.ZygoteInit.main(ZygoteInit.java:616)

Jasonchenlijian, 2017-12-10:
请问是用的哪个版本？我排查了一下代码，新版本在方法调用的地方进行了非空的处理，试试用新版本吧。

GoAheadComeOn, 2017-12-10:
你好我使用的2.1.1版本，那我更新下版本在测试下，谢谢

closed 2018-01-05 by Jasonchenlijian