Jasonchenlijian/FastBle #162: 有没有什么办法可以让onScanning一直执行

jhxiaoming, 2017-12-06:
public void onScanning(BleDevice bleDevice) {
  		// 扫描到一个符合扫描规则的BLE设备（UI线程）
      }
项目有一个需求，通过蓝牙广播来传递数据，有没有什么办法，可以让扫描的方法持续不断的执行？

Jasonchenlijian, 2017-12-07:
把扫描超时时间设置为0

jhxiaoming, 2017-12-08:
嗯，把扫描超时设置为0确实可以实现无限扫描，多谢大佬。还有一个问题。 onScanning(BleDevice bleDevice)这个方法，貌似是对设备的监听，每出现一个设备，就可以获取该设备的信息、信号强度、广播包数据。现在公司项目里面有这样一个需求：当硬件状态改变的时候就会发送广播，我在客户端要对广播包中的数据进行监听。 我本来的想法是不断的在后台扫描设备接受广播。但是无奈发现，在Andorid端， onScanning(BleDevice bleDevice)方法中，设备只能够扫描到一次。但是IOS端可以完成，请问是我使用的姿势不对吗？

jhxiaoming, 2017-12-08:
public interface LeScanCallback {
        /**
         * Callback reporting an LE device found during a device scan initiated
         * by the {@link BluetoothAdapter#startLeScan} function.
         *
         * @param device Identifies the remote device
         * @param rssi The RSSI value for the remote device as reported by the
         *             Bluetooth hardware. 0 if no RSSI value is available.
         * @param scanRecord The content of the advertisement record offered by
         *                   the remote device.
         */
        public void onLeScan(BluetoothDevice device, int rssi, byte[] scanRecord);
    }
我看了蓝牙的源码，发现这个onLeScan()确实是设备被发现以后的回调。现在只能硬件被触碰以后，会发送广播包，有什么办法可以实时监听这个广播包呢？

Jasonchenlijian, 2017-12-08:
你好，你上面提到的方法没有错，在这个回调方法里面执行你的业务需求是合适的。这是ble执行扫描的官方方法的一种，也是目前框架内用到的基础方法，不过框架在这个回调方法内做了一下过滤，如果是相同的设备只会回调一次onScanning。
我的建议是，第一，如果仅仅是你上面提到的这个简单的也许需求，其实没有必要用这个fastble框架，因为扫描太简单了，用了框架反而会制约你的发挥。第二，如果你实在想用框架，可以拷贝源码小幅修改一下。我有空也会修改增加这个方法回调。

jhxiaoming, 2017-12-08:
多谢指导！

jhxiaoming, 2017-12-08:
多谢指导！

Jasonchenlijian, 2017-12-10:
2.1.5版本已经增加了你的需求。

closed 2017-12-10 by Jasonchenlijian