Jasonchenlijian/FastBle #181: 连接稳定性

JiWenBo, 2017-12-22:
我FastBle 还有RxAndroidBLe都用过，但是一个明显的区别就是，RxAndrodBLe不管是连接的速度，还是连接成功率都要高一点。我们项目现在用的FastBle，扫描我们的设备10几秒才能扫描到，而且连接特别不顺畅，换了RxAndrodiBLe以后，顺畅度基本和IOS一样了。我想知道你在这个框架里做 了什么处理吗？为什么会出现这些问题。

Jasonchenlijian, 2017-12-22:
`如果是扫描部分出现有问题的话，我能想到的FastBle中使用的API21之前的`startLeScan`，其中并没有加入延迟或者阻塞的相关代码。你可以试一下原生的扫描方法是否需要这么长时间：

        BluetoothAdapter bluetoothAdapter = BluetoothAdapter.getDefaultAdapter();
        bluetoothAdapter.startLeScan(new BluetoothAdapter.LeScanCallback() {
            @Override
            public void onLeScan(BluetoothDevice device, int rssi, byte[] scanRecord) {

            }
        });

另外也可以尝试使用API21以上的其他原生扫描方法测试扫描速度，而后使用 `BleDevice convertBleDevice(ScanResult scanResult)`接着使用FastBle框架中的方法。

最后就是扫描的时候要区分回调方法 `onScanning`和 `onScanFinished`，以及配置超时时间。

至于连接不顺畅的问题，是否指的连接时间？假如排除设备因素的话，我能想到的可能就是autoConnect参数的配置或者其他。

JiWenBo, 2017-12-22:
我没有配置自动连接。连接的时候连接失败的情况确实较多，而且速度确实和rxAndroidble比起来有明显的差距。我们用fastble用户投诉比较多。无奈。您可以试一下rx那个，做一下比较

| |
冀文博
|
|
邮箱：jwb9325@163.com
|

签名由 网易邮箱大师 定制

在2017年12月22日 14:37，陈利健 写道：

如果是扫描部分出现有问题的话，我能想到的FastBle中使用的API21之前的startLeScan`，其中并没有加入延迟或者阻塞的相关代码。你可以试一下原生的扫描方法是否需要这么长时间：

    BluetoothAdapter bluetoothAdapter = BluetoothAdapter.getDefaultAdapter();
    bluetoothAdapter.startLeScan(new BluetoothAdapter.LeScanCallback() {
        @Override
        public void onLeScan(BluetoothDevice device, int rssi, byte[] scanRecord) {

        }
    });

另外也可以尝试使用API21以上的其他原生扫描方法测试扫描速度，而后使用 BleDevice convertBleDevice(ScanResult scanResult)接着使用FastBle框架中的方法。

最后就是扫描的时候要区分回调方法 onScanning和 onScanFinished，以及配置超时时间。

至于连接不顺畅的问题，是否指的连接时间？假如排除设备因素的话，我能想到的可能就是autoConnect参数的配置或者其他。

Jasonchenlijian, 2017-12-22:
好的，我会去试试。另外你使用的FastBle是哪个版本？

JiWenBo, 2017-12-26:
我用的还是1.0.6版本的

4what2do, 2018-01-03:
我用FASTBLE2.1.7比RxAndroidBLE1.4.3的连接稳定些，后者写数据的时候，总是经常莫名的断掉了

Jasonchenlijian, 2018-01-08:
考虑稳定性建议使用比较新的2.1.x 和2.2.x这几个版本，做了一些优化。

closed 2018-01-08 by Jasonchenlijian