Jasonchenlijian/FastBle #211: 蓝牙通过uuid过滤貌似没效果啊大兄弟

ruanbaojun1105, 2018-01-23:
UUID uuid1=UUID.fromString("00001800-0000-1000-8000-00805f9b34fb");
        UUID uuid2=UUID.fromString("00001801-0000-1000-8000-00805f9b34fb");
        UUID uuid3=UUID.fromString("49535343-fe7d-4ae5-8fa9-9fafd205e455");
        BleScanRuleConfig scanRuleConfig = new BleScanRuleConfig.Builder()
                .setServiceUuids(new UUID[]{uuid1,uuid2,uuid3})      // 只扫描指定的服务的设备，可选
                //.setDeviceName(true, names)   // 只扫描指定广播名的设备，可选
                //.setDeviceMac(mac)                  // 只扫描指定mac的设备，可选
                //.setAutoConnect(isAutoConnect)      // 连接时的autoConnect参数，可选，默认false
                .setScanTimeOut(10000)              // 扫描超时时间，可选，默认10秒
                .build();
        BleManager.getInstance().initScanRule(scanRuleConfig);

这样写有问题吗

Jasonchenlijian, 2018-01-23:
没问题

ruanbaojun1105, 2018-01-23:
写了过滤就直接不显示了，不加可以 ，确定uuid一串没有错，咋搞？

hisonce, 2018-05-04:
我的也是这个问题

ruanbaojun1105, 2018-05-21:
无法通过uuid过滤，望大神下个版本能修复或者在看下，谢了大神辛苦了，我也再看看

Jasonchenlijian, 2018-05-21:
@ruanbaojun1105 很抱歉，一直没有重点关注你这个问题，近期我会对你这个问题作出回答或者修改。

ruanbaojun1105, 2018-05-21:
@Jasonchenlijian 好的，谢了。

dilemma1212, 2019-08-21:
您好，请问您现在这个问题解决了吗

bulazhang, 2019-09-23:
今天测了一下，貌似是可以的

xiezuan, 2019-12-03:
Android 4.4 可以，手上Android 7 ,Android 9的设备不行

lxa520, 2019-12-07:
我测试了，还是不可以。

waylenZheng, 2020-03-19:
> 我测试了，还是不可以。

这个还没修复。。。。2年了

Liberations, 2020-05-13:
没救了 我也遇到这问题。

daoqilvzouzheqiao, 2021-07-02:
同样遇到了