Jasonchenlijian/FastBle #212: 请教如何订阅通知 notify，以及 write，read

BudinMilk, 2018-01-24:
以下是我的代码

public class OperationActivity extends AppCompatActivity implements Observer {

    public static final String KEY_DATA = "key_data";

    private BleDevice bleDevice;
    private BluetoothGattService bluetoothGattService;
    private BluetoothGattCharacteristic characteristic;
    private int charaProp;

    private Toolbar toolbar;
    private List<Fragment> fragments = new ArrayList<>();
    private int currentPage = 0;
    private String[] titles = new String[3];

    private static final String uuid_service = "6E400001-B5A3-F393-E0A9-E50E24DCCA9E";
    private static final String uuid_notify = "6E400003-B5A3-F393-E0A9-E50E24DCCA9E";
    private static final String uuid_write = "6E400002-B5A3-F393-E0A9-E50E24DCCA9E";
    private static final String uuid_read = "6E400003-B5A3-F393-E0A9-E50E24DCCA9E";

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);

        setContentView(R.layout.activity_operation);

        BleManager.getInstance().notify(
                bleDevice,
                uuid_service,
                uuid_notify,
                new BleNotifyCallback() {

                    @Override
                    public void onNotifySuccess() {
                        runOnUiThread(new Runnable() {
                            @Override
                            public void run() {
                                Log.d("TAG", "onNotifySuccess");
                            }
                        });
                    }

                    @Override
                    public void onNotifyFailure(final BleException exception) {
                        runOnUiThread(new Runnable() {
                            @Override
                            public void run() {
                                Log.d("TAG", "onNotifyFailure");
                            }
                        });
                    }

                    @Override
                    public void onCharacteristicChanged(byte[] data) {
                        runOnUiThread(new Runnable() {
                            @Override
                            public void run() {
                                Log.d("TAG", "onCharacteristicChanged");
                            }
                        });
                    }
                });

        BleManager.getInstance().write(
                bleDevice,
                uuid_service,
                uuid_write,
                HexUtil.hexStringToBytes("AABBCC"),
                new BleWriteCallback() {
                    @Override
                    public void onWriteSuccess(int current, int total, byte[] justWrite) {
                        // 发送数据到设备成功
                    }

                    @Override
                    public void onWriteFailure(BleException exception) {
                        // 发送数据到设备失败
                    }
                });

        BleManager.getInstance().read(
                bleDevice,
                uuid_service,
                uuid_read,
                new BleReadCallback() {
                    @Override
                    public void onReadSuccess(byte[] data) {
                        // 读特征值数据成功
                    }

                    @Override
                    public void onReadFailure(BleException exception) {
                        // 读特征值数据失败
                    }
                });

/*
        initData();
        initView();
        initPage();

        ObserverManager.getInstance().addObserver(this);
*/
    }

一直出现 onNotifyFailure，所以后面的 write，read 也没办法继续，不知道是哪里写错，请大神指导，谢谢！

Jasonchenlijian, 2018-01-24:
你这样写，bleDevice对象赋值了吗？是空吗？

BudinMilk, 2018-01-24:
感谢，我把 bleDevice 对象如何赋值之后成功了，再请问 write 哪里有错呢？

Jasonchenlijian, 2018-01-24:
write没错。但是，建议在 onNotifySuccess()结果里再去write。或者等notify方法执行完，延时几百毫秒进行write。

BudinMilk, 2018-01-24:
好的，感谢您的回复，谢谢！

closed 2018-01-24 by BudinMilk