Jasonchenlijian/FastBle #213: 请教何时呼叫 read 接收资料

BudinMilk, 2018-01-24:
您好，我目前可以成功发送数据到设备，但是不知道如何接收资料，我不晓得何时呼叫
read() 这个方法，是否有人能指点一下呢，感谢各位！

public class OperationActivity extends AppCompatActivity implements Observer {

    public static final String KEY_DATA = "key_data";

    private BleDevice bleDevice;
    private BluetoothGattService bluetoothGattService;
    private BluetoothGattCharacteristic characteristic;
    private int charaProp;

    private Toolbar toolbar;

    private static final String uuid_service = "6E400001-B5A3-F393-E0A9-E50E24DCCA9E";
    private static final String uuid_notify = "6E400003-B5A3-F393-E0A9-E50E24DCCA9E";
    private static final String uuid_write = "6E400002-B5A3-F393-E0A9-E50E24DCCA9E";
    private static final String uuid_read = "6E400003-B5A3-F393-E0A9-E50E24DCCA9E";

    private Button btnWrite;
    private TextView txt;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);

        setContentView(R.layout.activity_operation);

        initData();
        initView();

        txt = (TextView) findViewById(R.id.txt);

        btnWrite = (Button) findViewById(R.id.btnWrite);
        btnWrite.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                write();
            }
        });
        ObserverManager.getInstance().addObserver(this);
    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        BleManager.getInstance().clearCharacterCallback(bleDevice);
        ObserverManager.getInstance().deleteObserver(this);
    }

    @Override
    public void disConnected(BleDevice device) {
        if (device != null && bleDevice != null && device.getKey().equals(bleDevice.getKey())) {
            finish();
        }
    }

    private void initView() {
        toolbar = (Toolbar) findViewById(R.id.toolbar);
        setSupportActionBar(toolbar);
        getSupportActionBar().setDisplayHomeAsUpEnabled(true);
    }

    private void initData() {
        bleDevice = getIntent().getParcelableExtra(KEY_DATA);
        if (bleDevice == null)
            finish();
        mNotify();
    }

    public BleDevice getBleDevice() {
        return bleDevice;
    }

    public BluetoothGattService getBluetoothGattService() {
        return bluetoothGattService;
    }

    public void setBluetoothGattService(BluetoothGattService bluetoothGattService) {
        this.bluetoothGattService = bluetoothGattService;
    }

    public BluetoothGattCharacteristic getCharacteristic() {
        return characteristic;
    }

    public void setCharacteristic(BluetoothGattCharacteristic characteristic) {
        this.characteristic = characteristic;
    }

    public int getCharaProp() {
        return charaProp;
    }

    public void setCharaProp(int charaProp) {
        this.charaProp = charaProp;
    }

    private void read() {
        BleManager.getInstance().read(
                bleDevice,
                uuid_service,
                uuid_read,
                new BleReadCallback() {
                    @Override
                    public void onReadSuccess(final byte[] data) {
                        runOnUiThread(new Runnable() {
                            @Override
                            public void run() {
                                Log.d("TAG", data.toString());
                            }
                        });
                    }

                    @Override
                    public void onReadFailure(BleException exception) {
                        Log.d("TAG", "onReadFailure");
                    }
                });
    }

    private void write() {
        BleManager.getInstance().write(
                bleDevice,
                uuid_service,
                uuid_write,
                HexUtil.hexStringToBytes("AABBCC"),
                new BleWriteCallback() {
                    @Override
                    public void onWriteSuccess() {
                        Log.d("TAG", "onWriteSuccess");
                    }

                    @Override
                    public void onWriteFailure(BleException exception) {
                        Log.d("TAG", "onWriteFailure");
                    }
                });
    }

    private void mNotify() {
        try {
            Thread.sleep(200);
        } catch (InterruptedException e) {
            e.printStackTrace();
        }

        BleManager.getInstance().notify(
                bleDevice,
                uuid_service,
                uuid_notify,
                new BleNotifyCallback() {

                    @Override
                    public void onNotifySuccess() {
                        Log.d("TAG", "onNotifySuccess");
                    }

                    @Override
                    public void onNotifyFailure(final BleException exception) {
                        Log.d("TAG", "onNotifyFailure");
                    }

                    @Override
                    public void onCharacteristicChanged(byte[] data) {
                        Log.d("TAG", "onCharacteristicChanged");
                    }
                });
    }
}

Jasonchenlijian, 2018-01-27:
先确定硬件，找到硬件上接收数据的那个特征和接收方式

ruanbaojun1105, 2018-01-29:
先找到服务和特征，只要有就直接打开通知

BudinMilk, 2018-02-01:
感谢各位，我了解了，谢谢

closed 2018-02-01 by BudinMilk