Jasonchenlijian/FastBle #217: 分包传输文件过大时（30kb+），闪退

thearyong, 2018-01-31:
打印报堆栈异常：
``` shell
E/AndroidRuntime: FATAL EXCEPTION: Thread-1736
                                                   Process: com.aiiage.steam.mobile, PID: 22884
                                                   java.lang.StackOverflowError: stack size 1036KB
                                                       at com.clj.fastble.utils.BleLog.d(BleLog.java:52)
                                                       at com.clj.fastble.utils.SplitTransferUtil$1.onWriteSuccess(SplitTransferUtil.java:55)
                                                       at com.clj.fastble.bluetooth.BleConnector$ConnectorHandler.handleMessage(BleConnector.java:132)
                                                       at android.os.Handler.dispatchMessage(Handler.java:111)

```

排查到应该是SplitTransferUtil.write递归，不停的 newBleConnector 和 new BleWriteCallback对象导致。但目前没想到好的实现方式!

Jasonchenlijian, 2018-01-31:
谢谢你提出的问题，异常的原因看起来很明显，我会尽力修复这个问题。

Jasonchenlijian, 2018-02-02:
compile 'com.clj.fastble:FastBleLib:2.2.4' 解决这个问题。

thearyong, 2018-02-02:
@Jasonchenlijian 感谢，之前自己改的面目全非了😭。
刚换回你的实现方式，已验证解决之前的问题，感谢。

closed 2018-02-02 by thearyong