Jasonchenlijian/FastBle #221: How to achieve continuous RSSI read with FastBle?

DLeague9, 2018-02-01:
I connected with a Ble peripheral and after connection establish i called _BleManager.getInstance().readRssi(bd, rssiCallback);_ but its only called at once and break .
`scheduleTask4Rsii = Executors.newScheduledThreadPool(1);
        scheduleTask4Rsii.scheduleAtFixedRate(new Runnable() {
            public void run() {
                Log.i("Requesting Rssi: ", bd.getRssi()+"");
                BleManager.getInstance().readRssi(bd, rssiCallback);
            }
        }, 0, 1000, TimeUnit.MILLISECONDS);

final BleRssiCallback rssiCallback = new BleRssiCallback() {
        @Override
        public void onRssiFailure(BleException exception) {
            Log.i("Faild REading Rssi: ", exception.getDescription());
        }

        @Override
        public void onRssiSuccess(int rssi) {
            Log.i("Device Rssi: ", rssi + "");
        }
    };`
Please reply soon. thanks

liudeming1122, 2018-02-01:
I meeting the same problem

Jasonchenlijian, 2018-02-02:
compile 'com.clj.fastble:FastBleLib:2.2.4'

closed 2018-05-19 by Jasonchenlijian