Jasonchenlijian/FastBle #222: 华为p9 写入数据 102

yuchangfu, 2018-02-06:
华为p9手机写入数据 返回异常102

Jasonchenlijian, 2018-02-07:
错误描述是什么？

yuchangfu, 2018-02-07:
执行到BleConnector中的writeCharacteristic
`  public void writeCharacteristic(byte[] data, BleWriteCallback bleWriteCallback, String uuid_write) {
        if (data == null || data.length <= 0) {
            if (bleWriteCallback != null)
                bleWriteCallback.onWriteFailure(new OtherException("the data to be written is empty"));
            return;
        }

        if (mCharacteristic == null
                || (mCharacteristic.getProperties() & (BluetoothGattCharacteristic.PROPERTY_WRITE | BluetoothGattCharacteristic.PROPERTY_WRITE_NO_RESPONSE)) == 0) {
            if (bleWriteCallback != null)
                bleWriteCallback.onWriteFailure(new OtherException("this characteristic not support write!"));
            return;
        }

        if (mCharacteristic.setValue(data)) {
            handleCharacteristicWriteCallback(bleWriteCallback, uuid_write);
            if (!mBluetoothGatt.writeCharacteristic(mCharacteristic)) {
                writeMsgInit();
                if (bleWriteCallback != null)
                    bleWriteCallback.onWriteFailure(new OtherException("gatt writeCharacteristic fail"));
            }
        } else {
            if (bleWriteCallback != null)
                bleWriteCallback.onWriteFailure(new OtherException("Updates the locally stored value of this characteristic fail"));
        }
    }`

这个函数中的mBluetoothGatt.writeCharacteristic(mCharacteristic)返回的是false

Jasonchenlijian, 2018-02-07:
建议用测试APK测试向这个特征值write，排除代码写法问题。

yuchangfu, 2018-02-07:
测试apk写，返回的错误是一样的
BleException{code=102,description='gatt writeCharacteristic fail'}

lynn8570, 2019-01-10:
我也遇到
BleException{code=102,description='gatt writeCharacteristic fail'}
写消息有时候成功有时候失败

Sennerai, 2019-04-04:
同问，华为手机BleException{code=102,description='gatt writeCharacteristic fail'}，其他品牌手机正常

longzekai, 2019-11-23:
> 同问，华为手机BleException{code=102,description='gatt writeCharacteristic fail'}，其他品牌手机正常

我遇到的这跟问题，是内容过长，分包的时候，没有限速导致的异常，因为默认的发送间隔是0ms，这个值建议设置成300ms。

closed 2021-12-07 by Jasonchenlijian

DeveloperJx, 2024-01-26:
可以试试在华为手机上和蓝牙设备配对，再进行数据发送