Jasonchenlijian/FastBle #223: 20字节一包，请问很长字节分包发送中间有间隔时长吗

swallowsonny, 2018-02-07:
20字节一包，请问很长字节分包发送中间有间隔时长吗？

Jasonchenlijian, 2018-02-07:
没有

swallowsonny, 2018-02-07:
@Jasonchenlijian 如果没有间隔，硬件接收会掉包，所以这个分包就失去意义了，感觉有个20ms的间隔就行了

Jasonchenlijian, 2018-02-07:
每次写入一包数据后会等到写完响应后再发送下一包

hisonce, 2018-05-04:
对于写入无返回的特征自需要间歇150毫秒在写入否则报错。
if (Looper.myLooper() != null && Looper.myLooper() == Looper.getMainLooper()) {
                                        try {
                                            Thread.sleep(150L);
                                        } catch (InterruptedException e) {
                                            BleLog.w(e.getMessage());
                                        }
                                        write();
                                    }