Jasonchenlijian/FastBle #226: 偶尔会连接失败  state 133

yang-mr, 2018-02-22:
BleGattCallback：onConnectionStateChange 
                                                             status: 133
                                                             newState: 0
                                                             currentThread: 528

newtonker, 2018-03-09:
@yang-mr 我这边开发测试的过程中，在很多手机上都会出现这个133。很多手机上失败率可以达到30%，我的建议是多试reconnect，终究会连上的。

yang-mr, 2018-03-15:
我这边不行  要重启下被连的蓝牙设备

Jasonchenlijian, 2018-04-29:
133错误的连接失败偶尔会发生，在不同机型上出现频率会不一样。解决方法是重连。重连是通过直接对已知设备对象或已知设备mac地址直接调用connect方法，而不经过扫描。如果扫描可能会扫不到设备，而必须通过重启蓝牙的手段恢复，不建议。

zfz1120, 2018-05-24:
E/MainActivity: onConnectionFailure: FootKeyBleException { code=102, description='GATT discover services exception occurred!'}

 E/MainActivity: onConnectionFailure: FootKeyConnectException{gattStatus=133, bluetoothGatt=android.bluetooth.BluetoothGatt@42aad198} BleException { code=101, description='Gatt Exception Occurred! '}

yang-mr, 2018-06-07:
reConnect is work ok!

enix223, 2018-09-05:
FastBLE是使用`.disconnect`来断开与设备的链接，但是下面的post提到，如果我们需要断开后再重新连接相同的设备，就需要使用`.close`来断开链接。还有一个post提到，需要在UI线程来建立连接，也不妨试试。

解决方法：

1. 使用`mGatt.close()`来关闭gatt链接
2. 与设备建立连接的时候，在UI线程执行:

```java
// Create handler for main thread where mContext is application context
mHandler = new Handler(mContext.getMainLooper());
...
// Connect to BLE device from mHandler
mHandler.post(new Runnable() {
  @Override
  public void run() {
    mBTGatt = mBTDevice.connectGatt(mContext, false, mGattCallback);
  }
});
```
3. 如果你打算支持的最低API为23，可以指定transport为`LE`来调用连接gatt:

```java
cGatt.connectGatt(this, false, gattCallback, BluetoothDevice.TRANSPORT_LE);
```

实践证明，使用1，2，也不能完全解决GATT_ERROR 133的问题，加上3后，133问题出现的机率大大降低了。（大概30次连接和断开重复操作后，出现2次）

[引用]:
1. https://stackoverflow.com/questions/25330938/android-bluetoothgatt-status-133-register-callback
2. https://stackoverflow.com/questions/28894111/android-ble-gatt-error133-on-connecting-to-device

zsfzbc, 2018-10-14:
GattException{gattStatus=133} BleException { code=101, description='Gatt Exception Occurred! '}

zsfzbc, 2018-10-14:
搞了几天了一直报这个错: GattException{gattStatus=133} BleException { code=101, description='Gatt Exception Occurred! '}

GoldTest, 2020-12-25:
重连间隔设置的长一些 有助于解决此问题

stevenwsg, 2021-11-12:
系统支持的话， 重启蓝牙能好一些嘛

yugithubqi, 2025-05-30:
133 的底层系统原因到底是什么呢？不知道的话只能乱试。。。。