Jasonchenlijian/FastBle #230: 华为android8.0 蓝牙写入数据失败，返回102，

istarvip, 2018-03-02:
BleException{code=102,description='gatt writeCharacteristic fail'}

在android 4.4 、android 5.1 android 6.0 android 7.0 均可以写入数据，android 8.0 写入数据失败

Jasonchenlijian, 2018-03-02:
在Android 8.0系统上我使用piexl验证过write数据没有问题，请问你使用的手机型号？

istarvip, 2018-03-02:
华为mate 10 pro

istarvip, 2018-03-02:
参考 [连接](https://github.com/istarvip/FastBle-master)
[image]
[image]
[image]

上述代码在android 4.4 、android 5.1 android 6.0 android 7.0 均可以写入成功，在华为mate10 pro android 8.0 写入数据失败

istarvip, 2018-03-02:
@Jasonchenlijian

PyBird, 2018-03-13:
楼主问题解决了吗？同问

lyp82nlf, 2018-06-27:
@istarvip @Jasonchenlijian @PyBird 这个问题我测出来了，有些手机刚连接上蓝牙就写入数据，会获取不到服务。建议中间加一点延时再测试一下。

Sennerai, 2019-04-04:
同样遇到@istarvip 的问题

wenfengtou, 2019-08-14:
解决了吗？@istarvip