Jasonchenlijian/FastBle #232: 关于BLE Disconnect无效的问题

PrinceChou, 2018-03-05:
Jason，我遇到了一个关于BLE Disconnect无效的问题，请求你的帮助。
APP的逻辑是这样的，首先选择一个BLE设备然后进入控制页面进行连接，连接成功后在控制页面中可以控制和接收通知的操作。但在控制页面中有一个PopMenu，里面可以通过startActivityForResult打开搜索BLE设备页并选择一个新的设备进行连接（这里的控制页面不得已执行了onStop）。我设想的是在用户选择一个新的设备后将之前连接的那个设备进行断开操作。我在选择新的设备后搜索页将会finish掉，然后进入控制页面的onActivityResult方法，在这里我先disconnect，但始终无效。

Jasonchenlijian, 2018-03-05:
你好，PrinceChou，从你的功能需求上看，FastBle框架里并不会有限制你开展这个功能的地方。我猜想，你在“设备搜索页”finish的时候是否调用了一些清理操作？或者，你着重提到了“控制页面”的onStop，你做了什么操作吗？

PrinceChou, 2018-03-06:
我没有做什么操作，现在已经放弃用startActivityForResult了，直接在打开新的设备搜索页面的同时把控制页Finish掉了。选择新的设备后再重新打开控制页，这时候disconnect是可以了，但还是有一个遗留问题，就是当手机发送一个数据给设备，设备收到消息并回复后会再次调用BluetoothGatt: onClientConnectionState()、BluetoothGatt: discoverServices()和BluetoothGatt: onSearchComplete()由于这几个回调会影响我应用的一些功能，所以还是没能彻底解决。不知道为什么上面那几个回调在**第二次**连接成功（第一次不会）后收到设备通知时会再次回调。谢谢！