Jasonchenlijian/FastBle #235: 华为MT7-TL10 EMUI3.0 可以扫描到设备，但是无法连接

flymorn, 2018-03-22:
HUAWEI  MT7-TL10 EMUI3.0 系统，Android 4.4.2，API 19，可以正常扫描到设备，但是无法连接，包括官方的FastBLE.apk。

打开调试窗口，调用 BleManager.getInstance().connect 时就报错如下:
`03-22 15:55:42.910 31039-31039/com.test.app I/dalvikvm: Could not find method android.bluetooth.BluetoothDevice.connectGatt, referenced from method com.clj.fastble.bluetooth.BleBluetooth.connect
03-22 15:55:42.910 31039-31039/com.test.app W/dalvikvm: VFY: unable to resolve virtual method 329: Landroid/bluetooth/BluetoothDevice;.connectGatt (Landroid/content/Context;ZLandroid/bluetooth/BluetoothGattCallback;I)Landroid/bluetooth/BluetoothGatt;
03-22 15:55:42.915 31039-31039/com.test.app I/dalvikvm: Could not find method android.bluetooth.BluetoothGattCallback.onMtuChanged, referenced from method com.clj.fastble.bluetooth.BleBluetooth$1.onMtuChanged
03-22 15:55:42.915 31039-31039/com.test.app W/dalvikvm: VFY: unable to resolve virtual method 350: Landroid/bluetooth/BluetoothGattCallback;.onMtuChanged (Landroid/bluetooth/BluetoothGatt;II)V
`

提示找不到：
android.bluetooth.BluetoothDevice.connectGatt  和  android.bluetooth.BluetoothGattCallback.onMtuChanged  这2个方法。

如果找不到对应的方法，不是应该进入到 ：
public void onConnectFail(BleException exception) {}
里面吗？

请教各位。

flymorn, 2018-03-23:
补充：
红米1s手机 miui 7.3， 基于 Android 4.4.4
华为荣耀畅玩4X CHE1-CL10，基于 Android 4.4.4

同样问题

shuijianbaozi, 2018-03-23:
这些手机 连接的时候放在线程中试试

dllen, 2018-11-27:
@flymorn 这个解决了吗？