Jasonchenlijian/FastBle #236: onNotifyFailure:code=102  descriptor equals null

Anigod, 2018-03-26:
连接蓝牙以后，调用BleManager.getInstance().notify,报了这个错误，无论是延迟再调用还是直接调用都是,代码如下：
` public void onConnectSuccess(final BleDevice bleDevice, BluetoothGatt gatt, int status) {
                MyTag.logMe("onConnectSuccess");
                // 连接成功，BleDevice即为所连接的BLE设备（主线程）
                tv_info.postDelayed(new Runnable() {
                    @Override
                    public void run() {
                        //需要延迟几秒钟
                        BleManager.getInstance().notify(bleDevice, "0003cdd0-0000-1000-8000-00805f9b0131", "0003cdd1-0000-1000-8000-00805f9b0131", new BleNotifyCallback() {
                            @Override
                            public void onNotifySuccess() {
                                // 打开通知操作成功
                                MyTag.logMe("onNotifySuccess");
                            }

                            @Override
                            public void onNotifyFailure(BleException exception) {
                                // 打开通知操作失败
                                MyTag.logMe(exception.getCode() + " onNotifyFailure " + exception.getDescription());
                            }

                            @Override
                            public void onCharacteristicChanged(byte[] data) {
                                // 打开通知后，设备发过来的数据将在这里出现
                                MyTag.logMe("onCharacteristicChanged" + ByteTools.byteToString_16(data, true));
                            }
                        });
                    }
                },2000);
            }`

源码中报错的地方是：
`BluetoothGattDescriptor descriptor = characteristic.getDescriptor(formUUID(UUID_CLIENT_CHARACTERISTIC_CONFIG_DESCRIPTOR));
        if (descriptor == null) {
            notifyMsgInit();
            if (bleNotifyCallback != null)
                bleNotifyCallback.onNotifyFailure(new OtherException("descriptor equals null"));
            return false;
        } `

Anigod, 2018-03-26:
emmm不知道为什么插入的代码变成上面那个鬼样子了。。。

Anigod, 2018-03-26:
在7.0和5.1版本手机上测试均是如此，使用demoapk的连接也是报同样的异常，而在v1.2.1版本中是正常的

closed 2018-03-26 by Anigod

Anigod, 2018-06-25:
我后来发现,虽然是报错了,但是仍然是可以收到通讯的,所以直接用就好了,那个报错莫名奇妙但是不需要理他

MorningBreeze, 2018-06-25:
好的，因为我们的notify含有多个，所以，只能自己在收到通知自己处理，谢谢啦

longzekai, 2019-11-27:
这个异常确实有点不爽，希望解决的童鞋，给出解决方案。