Jasonchenlijian/FastBle #242: 开启扫描时卡顿!

lm11000, 2018-04-04:
Skipped 121 frames!  The application may be doing too much work on its main thread.
04-04 16:14:50.454 14150-14150/cn.askj.kotlincat I/BleScanCallbackImpl: onScanStarted: 扫描开启
每次开启扫描之前都会先卡顿1-2秒的时间,请问一下这是扫描情况呢?到底是库的问题还是我的代码问题呀?

lm11000, 2018-04-04:
比较奇怪的是在亮屏状态,,没有绘制UI后台运行的情况下,不会有卡顿现象,但是一旦亮屏,UI线程绘制时(有自定义控件存在),每次开启扫描界面都会出现明显卡顿(2-3秒).

Jasonchenlijian, 2018-04-04:
请问你在每个扫描回调方法里面都执行了哪些操作？

lm11000, 2018-04-04:
补充一下,我是按照很久以前的版本的demo仿造的,将蓝牙操作封装在service中,由service统一控制管理开启扫描.
我检查了一下扫描回调的代码.里面只有一个eventBus和一个logcat.并且我都将其注释尝试过了,依然出现了卡顿情况.

04-04 17:32:07.126 11689-11689/cn.askj.kotlincat I/Choreographer: Skipped 119 frames!  The application may be doing too much work on its main thread.
04-04 17:32:07.151 11689-11689/cn.askj.kotlincat I/BleScanCallbackImpl: onScanStarted: 扫描开启

卡顿发生在扫描之前.当我适量删减掉部分消耗较大的自定义控件时,卡顿现象的确存在缓解情况,但是我是不能够删掉这些自定义控件的啊= =

我的自定义UI控件是一个无限循环自动滚动的banner和一个仿雷达扫描的线程调度的自定义View,请问是否是因为资源分配的问题呢?FastBle这一块有相关的性能优化空间吗?

Jasonchenlijian, 2018-04-04:
你用的哪个版本的框架呢？

lm11000, 2018-04-04:
之前使用的是2.1.1.现在直接更新到了最新版本了.

我刚换了测试机发现了问题所在了.我以前写这个项目用的手机是小米5.(android6.0)进行调试的,没有问题.
现在使用的测试机是小米6(android 8.0).结果发生了卡顿,是否这意味着这是手机的问题?

lm11000, 2018-04-04:
不好意思打扰了...我确认是手机问题...妈的辣鸡米6,后台系统不知道在搞什么大量占用了我的资源导致的卡顿问题...我重启了手机,问题已经解决,感谢

closed 2018-04-04 by lm11000