Jasonchenlijian/FastBle #270: 内存泄漏  扫描回调导致

okmin3, 2018-05-18:
BleManager.getInstance().scan(new BleScanCallback()) 该回调无法移除，并且我在调用了 BleManager.getInstance().cancelScan(); 之后仍然会收到 onScanFinished 回调（我扫描使用的是无时间限制的，过了一段时间之后偶尔会回调该方法）

建议：
查看源码 BleScanCallback 回调的上层是 BleScanPresenter 回调。
我认为可将 BleScanCallback  作为BleScanPresenter  的成员变量。
最终由 BleManager->BleScanner->BleScanPresenter 移除该回调

Jasonchenlijian, 2018-05-19:
谢谢你的建议，我接下来会仔细考虑这方面。

closed 2018-05-19 by Jasonchenlijian

docwei2050, 2020-12-15:
然而并没有，还是有内存泄露

hjywyj, 2023-04-14:
是的，目前只能手动反射清掉
` fun destroy() {
        BleManager.getInstance().destroy()
        val mBleScanPresenter = ReflectHelper.getFieldValue<Any>(BleScanner.getInstance(), "mBleScanPresenter")
        if (mBleScanPresenter != null) {
            ReflectHelper.setFieldValue(mBleScanPresenter, "mBleScanPresenterImp", null)
        }
    }`

snxoriu, 2024-08-13:
这问题确实很严重,每个页面使用扫描都会有内存泄漏.而且使用框架的,可能都是新手,并不懂怎么解决.我也研究好久才知道具体的泄漏位置是回调方法,我使用的java,解决方法,反射清理:
try {
            // 获取BleScanner中的mBleScanPresenter字段
            Field mBleScanPresenterField = BleScanner.class.getDeclaredField("mBleScanPresenter");
            mBleScanPresenterField.setAccessible(true);//设置私有可访问
            Object mBleScanPresenter = mBleScanPresenterField.get(BleScanner.getInstance());

            // 获取BleScanPresenter中的mBleScanPresenterImp字段
            Field mBleScanPresenterImpField = BleScanPresenter.class.getDeclaredField("mBleScanPresenterImp");
            mBleScanPresenterImpField.setAccessible(true);//设置私有可访问

            // 将mBleScanPresenterImp字段设置为null
            mBleScanPresenterImpField.set(mBleScanPresenter, null);

            //最后恢复私有
            mBleScanPresenterField.setAccessible(false);
            mBleScanPresenterImpField.setAccessible(false);

        } catch (Exception e) {
            e.printStackTrace();
        }
BleManager.getInstance().removeNotifyCallback(bleDevice0, UUID_Notify);
        BleManager.getInstance().removeMtuChangedCallback(bleDevice0);
        BleManager.getInstance().removeConnectGattCallback(bleDevice0);
}