Jasonchenlijian/FastBle #274: UUID过滤能否添加两个设备的主UUID过滤

HuaZaiWuDi, 2018-05-24:
意思就是我想扫描到两种设备。通过UUID过滤设置。添加两个设备的UUID。实现同时扫描两种设置

xizirumeng, 2018-06-20:
你能通过UUID过滤设备么？我一旦加了UUID过滤条件就搜不到设备了

Jasonchenlijian, 2018-06-21:
uuid是与的条件，而不是或

fengchuiyeluo, 2019-03-13:
不要设置过滤条件，然后在扫描回调onScanning中bleDevice.getScanRecord()获取BLE的广播数据，之后对该数据进行解析获取serviceUuid就能自己判断要哪个设备了

fengchuiyeluo, 2019-03-13:
`
import android.bluetooth.le.ScanRecord;
import android.os.ParcelUuid;
import android.support.annotation.Nullable;
import android.util.ArrayMap;
import android.util.Log;
import android.util.SparseArray;

import java.nio.ByteBuffer;
import java.nio.ByteOrder;
import java.util.ArrayList;
import java.util.Arrays;
import java.util.List;
import java.util.Map;
import java.util.UUID;

/**
 * Created by cfd on 2019/3/13.
 */
public class ScanRecordUtils {
    private static final String TAG = "ScanRecordUtils";

    // The following data type values are assigned by Bluetooth SIG.
    // For more details refer to Bluetooth 4.1 specification, Volume 3, Part C, Section 18.
    private static final int DATA_TYPE_FLAGS = 0x01;
    private static final int DATA_TYPE_SERVICE_UUIDS_16_BIT_PARTIAL = 0x02;
    private static final int DATA_TYPE_SERVICE_UUIDS_16_BIT_COMPLETE = 0x03;
    private static final int DATA_TYPE_SERVICE_UUIDS_32_BIT_PARTIAL = 0x04;
    private static final int DATA_TYPE_SERVICE_UUIDS_32_BIT_COMPLETE = 0x05;
    private static final int DATA_TYPE_SERVICE_UUIDS_128_BIT_PARTIAL = 0x06;
    private static final int DATA_TYPE_SERVICE_UUIDS_128_BIT_COMPLETE = 0x07;
    private static final int DATA_TYPE_LOCAL_NAME_SHORT = 0x08;
    private static final int DATA_TYPE_LOCAL_NAME_COMPLETE = 0x09;
    private static final int DATA_TYPE_TX_POWER_LEVEL = 0x0A;
    private static final int DATA_TYPE_SERVICE_DATA_16_BIT = 0x16;
    private static final int DATA_TYPE_SERVICE_DATA_32_BIT = 0x20;
    private static final int DATA_TYPE_SERVICE_DATA_128_BIT = 0x21;
    private static final int DATA_TYPE_MANUFACTURER_SPECIFIC_DATA = 0xFF;

    public static final ParcelUuid BASE_UUID =
            ParcelUuid.fromString("00000000-0000-1000-8000-00805F9B34FB");
    /**
     * Length of bytes for 16 bit UUID
     */
    public static final int UUID_BYTES_16_BIT = 2;
    /**
     * Length of bytes for 32 bit UUID
     */
    public static final int UUID_BYTES_32_BIT = 4;
    /**
     * Length of bytes for 128 bit UUID
     */
    public static final int UUID_BYTES_128_BIT = 16;

    // Flags of the advertising data.
    private final int mAdvertiseFlags;

    @Nullable
    private final List<ParcelUuid> mServiceUuids;

    private final SparseArray<byte[]> mManufacturerSpecificData;

    private final Map<ParcelUuid, byte[]> mServiceData;

    // Transmission power level(in dB).
    private final int mTxPowerLevel;

    // Local name of the Bluetooth LE device.
    private final String mDeviceName;

[... 173 lines cut ...]
        // 128_bit_value = uuid * 2^96 + BASE_UUID
        long shortUuid;
        if (length == UUID_BYTES_16_BIT) {
            shortUuid = uuidBytes[0] & 0xFF;
            shortUuid += (uuidBytes[1] & 0xFF) << 8;
        } else {
            shortUuid = uuidBytes[0] & 0xFF;
            shortUuid += (uuidBytes[1] & 0xFF) << 8;
            shortUuid += (uuidBytes[2] & 0xFF) << 16;
            shortUuid += (uuidBytes[3] & 0xFF) << 24;
        }
        long msb = BASE_UUID.getUuid().getMostSignificantBits() + (shortUuid << 32);
        long lsb = BASE_UUID.getUuid().getLeastSignificantBits();
        return new ParcelUuid(new UUID(msb, lsb));
    }

    /**
     * Returns a list of service UUIDs within the advertisement that are used to identify the
     * bluetooth GATT services.
     */
    public List<ParcelUuid> getServiceUuids() {
        return mServiceUuids;
    }
}
`

fengchuiyeluo, 2019-03-13:
上面代码是提取自源码的广播解析工具类，`ScanRecordUtils scanRecordUtils = ScanRecordUtils.parseFromBytes(bleDevice.getScanRecord());
                List<ParcelUuid> serviceUuids = scanRecordUtils.getServiceUuids();`

senonwx, 2019-05-05:
我也遇到同样的问题 添加过滤条件之后搜索不到ble设备   尝试采用
ScanRecordUtils scanRecordUtils = ScanRecordUtils.parseFromBytes(bleDevice.getScanRecord()); List<ParcelUuid> serviceUuids = scanRecordUtils.getServiceUuids();
发现ScanRecordUtils没有呢？

BestCoderXQX, 2019-05-16:
@senonwx  上面的意思是自己吧源码中的一个类搞出来当做自己的类 然后根据这个类就可以得到设备的uuid 然后根据这个uuid自己去筛选