Jasonchenlijian/FastBle #276: 连接断开后为啥要在列表删除设备

zfz1120, 2018-05-25:
如果不删除，显示的还是连接状态，查询状态也是连接成功的，实际调用了断开函数。。
没明白。。

zfz1120, 2018-06-04:
@Jasonchenlijian 

APP断主动断开。再次点击item，识别为连接状态。调用disconnect，情况如下：
  
 public void disconnect(BleDevice bleDevice) {
        if (multipleBluetoothController != null) {
            multipleBluetoothController.disconnect(bleDevice);
        }
    }

    public synchronized void disconnect(BleDevice bleDevice) {
        if (isContainDevice(bleDevice)) {
            getBleBluetooth(bleDevice).disconnect();
        }
    }
 isContainDevice直接判定无该设备。。。再也无法进入断开逻辑。。。
而连接状态则永久为luetoothProfile.STATE_CONNECTED;

Jasonchenlijian, 2018-06-04:
isContainDevice结果为false吗？为false的情况有两种，一种是未连接过，一种是收到断开回调，然后调用removeBleBluetooth方法，你检查一下。

zfz1120, 2018-06-05:
还有个问题。。。设置最大连接数为5、、、连接第4个的时候，返回超时，重连返回133、、、是框架写死了最大为3么？？ @Jasonchenlijian

Jasonchenlijian, 2018-06-08:
并没有写死最大连接数，超时可以重连。