Jasonchenlijian/FastBle #283: 获取到多个UUID，用哪一个？

ZhanRu, 2018-06-14:
onGetService/onGetCharacteristic/onGetDescriptor，这三个方法返回了很多UUID，不知道我用哪个进行通讯呢？
打印结果如下：
---------------------------------------------
 discoverServices() - device: F0:B5:D1:A6:8E:72
 onGetService() - Device=F0:B5:D1:A6:8E:72 UUID=0000ffe0-0000-1000-8000-00805f9b34fb
 onGetService() - Device=F0:B5:D1:A6:8E:72 UUID=0000fee7-0000-1000-8000-00805f9b34fb
 onGetService() - Device=F0:B5:D1:A6:8E:72 UUID=00001800-0000-1000-8000-00805f9b34fb
 onGetService() - Device=F0:B5:D1:A6:8E:72 UUID=00001801-0000-1000-8000-00805f9b34fb
 onGetCharacteristic() - Device=F0:B5:D1:A6:8E:72 UUID=0000ffe1-0000-1000-8000-00805f9b34fb
 onGetCharacteristic() - Device=F0:B5:D1:A6:8E:72 UUID=0000ffe2-0000-1000-8000-00805f9b34fb
 onGetCharacteristic() - Device=F0:B5:D1:A6:8E:72 UUID=0000fec7-0000-1000-8000-00805f9b34fb
 onGetCharacteristic() - Device=F0:B5:D1:A6:8E:72 UUID=0000fec9-0000-1000-8000-00805f9b34fb
 onGetCharacteristic() - Device=F0:B5:D1:A6:8E:72 UUID=0000fec8-0000-1000-8000-00805f9b34fb
 onGetCharacteristic() - Device=F0:B5:D1:A6:8E:72 UUID=00002a00-0000-1000-8000-00805f9b34fb
 onGetCharacteristic() - Device=F0:B5:D1:A6:8E:72 UUID=00002a01-0000-1000-8000-00805f9b34fb
 onGetCharacteristic() - Device=F0:B5:D1:A6:8E:72 UUID=00002a02-0000-1000-8000-00805f9b34fb
 onGetCharacteristic() - Device=F0:B5:D1:A6:8E:72 UUID=00002a03-0000-1000-8000-00805f9b34fb
 onGetCharacteristic() - Device=F0:B5:D1:A6:8E:72 UUID=00002a04-0000-1000-8000-00805f9b34fb
 onGetCharacteristic() - Device=F0:B5:D1:A6:8E:72 UUID=00002a05-0000-1000-8000-00805f9b34fb
 onGetDescriptor() - Device=F0:B5:D1:A6:8E:72 UUID=00002902-0000-1000-8000-00805f9b34fb
 onGetDescriptor() - Device=F0:B5:D1:A6:8E:72 UUID=00002901-0000-1000-8000-00805f9b34fb
 onGetDescriptor() - Device=F0:B5:D1:A6:8E:72 UUID=00002902-0000-1000-8000-00805f9b34fb
 onGetDescriptor() - Device=F0:B5:D1:A6:8E:72 UUID=00002901-0000-1000-8000-00805f9b34fb
 onGetDescriptor() - Device=F0:B5:D1:A6:8E:72 UUID=00002902-0000-1000-8000-00805f9b34fb
 onGetDescriptor() - Device=F0:B5:D1:A6:8E:72 UUID=00002902-0000-1000-8000-00805f9b34fb

ZhanRu, 2018-06-14:
哈哈，我搞懂了，IOS上用LightBlue检测一下，就有了。

NewbieFish, 2019-05-22:
怎么操作的呢，能不能详细说一下呢，我也是获取了很多UUID（一个特殊UUID），然后我拿这个特殊的服务UUID，去过滤设备，就搜不到了

ZhanRu, 2019-05-22:
> 怎么操作的呢，能不能详细说一下呢，我也是获取了很多UUID（一个特殊UUID），然后我拿这个特殊的服务UUID，去过滤设备，就搜不到了（不是通讯，是搜索蓝牙的过程）

IOS在应用市场下载LightBlue，这个软件打开后，可以看到现在用的哪个UUID连接的设备