Jasonchenlijian/FastBle #292: 指定设备名或设备MAC地址的话扫描不到设备。

zkpeng, 2018-06-29:


Jasonchenlijian, 2018-06-29:
可以展示一下相关代码

zkpeng, 2018-06-29:
setScanRule();//设置扫描规则
startScan();//开始扫描

private void setScanRule() {
//        UUID[] serviceUuids = {
//                Constants.UUID_SERVICE};
        BleScanRuleConfig scanRuleConfig = new BleScanRuleConfig.Builder()
//                .setServiceUuids(serviceUuids)      // 只扫描指定的服务的设备，可选
                .setDeviceName(false, "62220202")   // 只扫描指定广播名的设备，可选
//                .setDeviceMac("5D:F0:D8:5A:BE:C3")                  // 只扫描指定mac的设备，可选
//                .setAutoConnect(true)      // 连接时的autoConnect参数，可选，默认false
                .setScanTimeOut(10000)              // 扫描超时时间，可选，默认10秒
                .build();
        BleManager.getInstance().initScanRule(scanRuleConfig);
}

private void startScan() {
        BleManager.getInstance().scan(new BleScanCallback() {
            @Override
            public void onScanStarted(boolean success) {
                showShortToast("开始扫描。。。");
            }

            @Override
            public void onLeScan(BleDevice bleDevice) {
                super.onLeScan(bleDevice);
            }

            @Override
            public void onScanning(BleDevice bleDevice) {
                if (!TextUtils.isEmpty(bleDevice.getName())) {
                    showShortToast("scanned device--->"+bleDevice.getName());
                }
            }

            @Override
            public void onScanFinished(List<BleDevice> scanResultList) {
                if (scanResultList == null || scanResultList.size() == 0) {
                    showShortToast("未扫描到设备");
                }
            }
        });
}

Jasonchenlijian, 2018-06-29:
你用Demo apk 扫描一下，看你所需要连接的设备的蓝牙名称是否就是“62220202”。

zkpeng, 2018-06-29:
是这个名字

zkpeng, 2018-06-29:
能搜到，但我看不出我的程序哪有问题。。。

zkpeng, 2018-06-29:
好了，我改成传入名字数组就好了

zkpeng, 2018-06-29:
好了，我改成传入名字数组就好了 谢谢

closed 2018-06-29 by zkpeng