Jasonchenlijian/FastBle #309: 不开启蓝牙进行扫描后带来的问题

Symous, 2018-08-04:
当我们不启动蓝牙，但是仍然调用扫描方法的话，当调用bleManager.cancelScan()后，会有如下报错信息：
Attempt to invoke virtual mehotd 'boolean android.os.HandlerThread.quit()' on a null object reference.
当然我可以通过自己判断蓝牙是否开启等方法来决定是否调用cancelScan。 :)

Jasonchenlijian, 2018-08-06:
谢谢你的反馈

JustTrust, 2019-07-26:
Faced same issue
com.myApp E/AndroidRuntime: FATAL EXCEPTION: main
java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.os.HandlerThread.quit()' on a null object reference
        at com.clj.fastble.scan.BleScanPresenter.notifyScanStopped(BleScanPresenter.java:207)
        at com.clj.fastble.scan.BleScanner.stopLeScan(BleScanner.java:131)
        at com.clj.fastble.BleManager.cancelScan(BleManager.java:385)
        at myApp.bluetooth.BluetoothLowEnergy.stopObservingBlDevices(BluetoothLowEnergy.kt:94)