Jasonchenlijian/FastBle #313: bleDevice.getName  return null

wxh12138, 2018-08-10:
bleDevice.getName return null

Jasonchenlijian, 2018-08-20:
说明该设备的名字未设置。外设一般有两处名称可供参考，这两个名称均可以通过扫描得到，一个是attDevice Name，也就是bleDevice.getName。另一个是Local Name，可以通过解析广播的数据包得到`byte[] scanRecord`，解析方法网上可以查的到，如果找不到我可以贴出来。一般来说这两个名称是设置的一样的，但实际情况请与外围硬件那边沟通确认。

lilingdan, 2018-08-31:
偶尔连接成功之后gatt.getDevice().getName(); 获取的设备名称是 null ,大多数情况是可以获取设备名的，扫描也有设备名

lilingdan, 2018-09-08:
在demo 中也出现了连接成功后设备名是null 的情况

lilingdan, 2018-09-08:
我扫描到的设备是有名字的，连接完成大部分的时候也是有name的，但是出现过毫秒级连接的情况获取设备名就是NUll

Jasonchenlijian, 2018-09-09:
请问毫秒级连接是什么意思？

lilingdan, 2018-09-09:
就是在1秒之内就连接上了

lilingdan, 2018-09-09:
连接成功但是name 为 null 的情况是v1.0.0 出现的，2.3.2 暂时没有出现，发现的另一个问题是，连接多次之后discoverServices 的回调大概要30s 才会上来，若等待则等待时间过长， 若不等待做超时强行disconnect 会出现，disconnect 的回调上报之后设备和手机又连接上了而且 ，这个 gatt在内部和列表里面未保存只能通过关闭手机蓝牙的方式来释放它

lilingdan, 2018-09-18:
有一种情况，扫描的时候有名字，但是通过扫二维码connect(mac)（二维码中有mac 地址，通过mac 地址),   偶尔会出现BleDevice 的name 是null 的情况