Jasonchenlijian/FastBle #323: 使用mac地址连接蓝牙，在还没连接上的时候怎么取消连接？

vae01, 2018-09-11:
使用mac地址连接蓝牙，设置重连次数。那如果在还没有连上的时候如何取消蓝牙连接的这个动作？本来是打算用
if (mBleManager.getScanSate().equals(BleScanState.STATE_SCANNING))
                        mBleManager.cancelScan();
这种方式的，但是在连接的回调里打log发现mBleManager.getScanSate() 一直是STATE_IDLE，然后直接去掉判断，走取消扫描mBleManager.cancelScan();的话就报错了 ：
E/AndroidRuntime(30313): java.lang.NullPointerException: Attempt to invoke virtual method 'boolean android.os.HandlerThread.quit()' on a null object reference
E/AndroidRuntime(30313): 	at com.clj.fastble.scan.BleScanPresenter.notifyScanStopped(BleScanPresenter.java:207)
E/AndroidRuntime(30313): 	at com.clj.fastble.scan.BleScanner.stopLeScan(BleScanner.java:131)
E/AndroidRuntime(30313): 	at com.clj.fastble.BleManager.cancelScan(BleManager.java:383)
不知道还有没有其他方式可以停止，蓝牙连接。

kanshenmekan, 2018-12-06:
解决了吗，一样的问题

Jasonchenlijian, 2018-12-31:
谢谢反馈，已在2.3.3版本中修复重连次数不重置为题。

closed 2018-12-31 by Jasonchenlijian