Jasonchenlijian/FastBle #327: HexUtil工具类字节与十六进制字符串的转换存在问题

coderJohnZhang, 2018-10-08:
作者，您好！

我在使用您的库的时候，发现如果我写入数据的字符串长度是奇数的话，解密后拿到的字符串会丢失：

比如我输入abcde，输出却是abcd:

` String str = "abcde";`
`System.out.println("origin string = " + str);`
`byte[] bytes = hexStringToBytes(str);`
`System.out.println("string to bytes = " + Arrays.toString(bytes));`
`str = formatHexString(bytes, false);`
`System.out.println("bytes to string = " + str);`

[image]

LiuZhiPeng1214, 2018-10-26:
作者你好！

我在使用您的库的时候发现HexUtil 中hexStringToBytes 没有加空格判断，导致数据错乱，建议加去掉空格的逻辑

Jasonchenlijian, 2018-12-31:
好，谢谢

closed 2018-12-31 by Jasonchenlijian

lqb12172, 2022-06-18:
在网上找的一些代码，拼凑一下修复这个bug
//string转byte数组
public static byte[] hexStringToBytes(String hexString) {
        /*if (hexString == null || hexString.equals("")) {
            return null;
        }
        hexString = hexString.trim();
        hexString = hexString.toUpperCase();
        int length = hexString.length() / 2;
        char[] hexChars = hexString.toCharArray();
        byte[] d = new byte[length];
        for (int i = 0; i < length; i++) {
            int pos = i * 2;
            d[i] = (byte) (charToByte(hexChars[pos]) << 4 | charToByte(hexChars[pos + 1]));
        }
        return d;*/

 if (hexString == null || hexString.equals("")) {
            return null;
        }
        //1.String数据转换成hexString
        String hexStr = HexUtil.str2HexStr(hexString);
        //2.hexString转换成byte数组
        byte[] hexBytes = HexUtil.hexString2Bytes(hexStr);
        if (hexBytes != null) {
            return hexBytes;
        } else {
            return null;
        }
    }

//byte数组转string
public static String formatHexString(byte[] data, boolean addSpace) {
        /*if (data == null || data.length < 1)
            return null;
        StringBuilder sb = new StringBuilder();
        for (int i = 0; i < data.length; i++) {
            String hex = Integer.toHexString(data[i] & 0xFF);
            if (hex.length() == 1) {
                hex = '0' + hex;
            }
            sb.append(hex);
            if (addSpace)
                sb.append(" ");
        }
        return sb.toString().trim();*/

        if (data == null || data.length < 1)
            return null;
        //1.byte数组转换成hexString
        String bytesToHexString = HexUtil.bytes2HexString(data);
        //2.hexString转换成String
        if (bytesToHexString != null) {
            return HexUtil.hexStr2Str(bytesToHexString);
        } else {
            return null;
        }
    }

/**
     * 字符串转换成为16进制(无需Unicode编码)
     * @param str
     * @return
     */
    public static String str2HexStr(String str) {
        char[] chars = "0123456789ABCDEF".toCharArray();
        StringBuilder sb = new StringBuilder("");
        byte[] bs = str.getBytes();
        int bit;
        for (int i = 0; i < bs.length; i++) {
            bit = (bs[i] & 0x0f0) >> 4;
            sb.append(chars[bit]);
            bit = bs[i] & 0x0f;
            sb.append(chars[bit]);
            // sb.append(' ');
        }
        return sb.toString().trim();
    }

    /**
     * 16进制直接转换成为字符串(无需Unicode解码)
     * @param hexStr
     * @return
     */
    public static String hexStr2Str(String hexStr) {
        String str = "0123456789ABCDEF";
        char[] hexs = hexStr.toCharArray();
        byte[] bytes = new byte[hexStr.length() / 2];
        int n;
        for (int i = 0; i < bytes.length; i++) {
            n = str.indexOf(hexs[2 * i]) * 16;
            n += str.indexOf(hexs[2 * i + 1]);
            bytes[i] = (byte) (n & 0xff);
        }
        return new String(bytes);
    }

    /**
     * 将byte数组转换成string类型表示
     * @param src
     * @return
     */
    public static String bytes2HexString(byte[] src) {
        StringBuilder builder = new StringBuilder();
        if (src == null || src.length <= 0) {
            return null;
        }
        String hv;
        for (int i = 0; i < src.length; i++) {
            // 以十六进制（基数 16）无符号整数形式返回一个整数参数的字符串表示形式，并转换为大写
            hv = Integer.toHexString(src[i] & 0xFF).toUpperCase();
            if (hv.length() < 2) {
                builder.append(0);
            }
            builder.append(hv);
        }

        return builder.toString();
    }

    /**
     * 将Hex String转换为Byte数组
     *
     * @param hexString the hex string
     * @return the byte [ ]
     */
    public static byte[] hexString2Bytes(String hexString) {
        if (TextUtils.isEmpty(hexString)) {
            return null;
        }
        hexString = hexString.toLowerCase();
        final byte[] byteArray = new byte[hexString.length() >> 1];
        int index = 0;
        for (int i = 0; i < hexString.length(); i++) {
            if (index  > hexString.length() - 1) {
                return byteArray;
            }
            byte highDit = (byte) (Character.digit(hexString.charAt(index), 16) & 0xFF);
            byte lowDit = (byte) (Character.digit(hexString.charAt(index + 1), 16) & 0xFF);
            byteArray[i] = (byte) (highDit << 4 | lowDit);
            index += 2;
        }
        return byteArray;
    }