Jasonchenlijian/FastBle #334: 关于 setCharacteristicNotification 请教.

Songsayit, 2018-10-22:
你好. 
  我想请教一下. 
BluetoothGattCharacteristic characteristic,
        BluetoothGattDescriptor descriptor = characteristic.getDescriptor(formUUID(UUID_CLIENT_CHARACTERISTIC_CONFIG_DESCRIPTOR));
这里的 UUID_CLIENT_CHARACTERISTIC_CONFIG_DESCRIPTOR , 你是怎么想到的呢?

 因为 我反编译 TI 的 ble apk 发现他获取description是通过下面方式:
BluetoothGattDescriptor clientConfig = characteristic.getDescriptor(characteristic.getUuid());

而我用的方法在我经手的2个BLE模块都是可以测试通过, 但是在客户新的BLE模块就需要使用您提供的方法, 这个让我很惊讶.
虽然协议里是规定了2902 是client description.

Jasonchenlijian, 2018-10-23:
2902是协议固定的哦。
Bluetooth SIG定义了许多标准的，设备可以去兼容，也可以自定义。而2902就是 GATT Descriptors中的Client Characteristic Configuration。比如2903是 Server Characteristic Configuration

Songsayit, 2018-10-23:
谢谢陈工的回复;
我看你博客里有些朋友提问就是他们的 setCharacteristicNotification  失败.
所以我猜想, 这个是由于BLE 设备多样性导致的, 
可以考虑在 setCharacteristicNotification  增加一些兼容性代码;
比如 ::
BluetoothGattDescriptor descriptor = characteristic.getDescriptor(formUUID(UUID_CLIENT_CHARACTERISTIC_CONFIG_DESCRIPTOR));
if(descriptor  == null) {
//这个可能是BLE模块那边在定义descrptor的时候, 把2902这个直接去掉了;
//使用他自己的characteristic的descrptor
 descriptor = characteristic.getDescriptor(characteristic.getUuid());
}

if(descriptor  == null) {
 //BLE模块那边忘记定义了. 导致 无法 setCharacteristicNotification  
}

Jasonchenlijian, 2018-10-23:
好的，你的建议 我思考一下